nightscout/AndroidAPS #2343: [DEV][NSC V3] Slow sync of data when loading for the first time

jbr7rr, 2023-01-12:
Build: https://github.com/nightscout/AndroidAPS/commit/6fb8339238059d17715f9aa9cdf3b9f86f479708-2023.01.12 (dev-g) fullrelease
Logs: [AndroidAPS_2023-01-12_JB_slow_loading.zip](https://github.com/nightscout/AndroidAPS/files/10404803/AndroidAPS_2023-01-12_JB_slow_loading.zip)
NS Version: 14.2.6

As discussed in discord, the data sync seems to slow down around 27-10, and no data comes available in NS.

Der-Schubi, 2023-01-12:
My status shows "Working" instead of "Not connected", but its SYNCING BG values for eternity.
[Uploading AndroidAPS._2023-01-12_18-39-41_.220.zip…]()
I'm also using NS 14.2.6

databoy2k, 2023-01-12:
My status is "not connected" and I'm having the eternal download of 500 SVGs from 10/27/22. 14.2.6, "admin" role on NS Token, built to 87ca7ef.

Der-Schubi, 2023-01-13:
> not traceable from log. I need config+pass via PM

Sent it on Discord.

MilosKozak, 2023-01-13:
fixed by https://github.com/nightscout/AndroidAPS/commit/378002df336294fe31495c50002ef139e806e2a4

closed 2023-01-13 by MilosKozak