nightscout/AndroidAPS #2347: EOPatch and AAPS Erros

Viewquest, 2023-01-15:
Hello Milos, I noticed yesterday that after changing the EOPatch pump I could no longer access the configuration of the DYNISF. AAPS kept crashing. Today I switched to the virtual pump. I wanted to test if I could work with that in the configuration. With this setting, everything could operate normally. I believe that an error has crept in with the EOPatch. Currently I have the DEV G 378002df33 installed. It would be great if the error can be detected and fixed.---------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new
[image]

MilosKozak, 2023-01-15:
i cannot replicate the issue. pm me crashing config and master pass

Viewquest, 2023-01-16:
I will set a new pod tonight and then email you the settings and master password.

Many greetings
Lothar

Viewquest, 2023-01-16:
Hello milos,

Here comes the data and my master password, as requested. I changed the EOPatch and went to the Settings button after the start. AAPS crashed again. Also, before that I built and installed the latest APK in Android Studio
.

Mein Masterkennwort: 31121957

MilosKozak, 2023-01-17:
resolved by https://github.com/nightscout/AndroidAPS/commit/fa2b9bddcceee6c9dc071e6055d8ff365d5b6e0b

closed 2023-01-17 by MilosKozak