nightscout/AndroidAPS #2349: Keine Notizen zu Nightscout

jfulge, 2023-01-15:
Einen wunderschönen guten Abend liebe Entwickler. Ich wünsche noch alles erdenklich Gute im neuen Jahr. 
Ich weiß jetzt nicht, wie ich beginnen soll. 
Ich benutze AAPS 3.1.0.2 unter Android 11. In der letzten Zeit stelle ich fest, dass die Notizen, die ich in AAPS unter dem Button der Kohlenhydrate eintrage, nicht zu  #1279  übertragen werden. In den Behandlungen, Benutzereingaben werden sie aufgeführt. Aber in Nightscout kommen sie nicht an. 
Ich hoffe, ihr könnt mir helfen? 
Viele Dank für eure Hilfe und Unterstützung. 
Viele Grüße von Jörg Fulge 
[image]
[image]

maheini, 2023-01-15:
Hi @jfulge 

Könntest du den Text & Titel auf Englisch verfassen? Viele Entwickler sind Englischsprachig, aber nicht deutsch. Ich habe den Inhalt kurz übersetzt, du musst also nur auf bearbeiten klicken & einfügen:

<details>
    <summary>Titel</summary>
No notes about Nightscout in Android APS
</details>
<details>
    <summary>Inhalt</summary>
Have a wonderful good evening dear developers. I wish you all the best in the new year.
I don't know how to start now.
I'm using AAPS 3.1.0.2 on Android 11. Lately I've noticed that the notes I put in AAPS under the carbs button don't transfer to #1279. In the treatments, user input they are listed. But they don't arrive in Nightscout.
I hope you can help me?
Many thanks for your help and support.
Greetings from Joerg Fulge
</details>

maheini, 2023-01-15:
What does your queue states on the Nightscout tab within AndroidAPS? Are there any treatments in queue?

Which Nighscout version are you using?

jfulge, 2023-01-16:
Have a wonderful good evening dear developers. I wish you all the best in the new year. I don't know how to start now. I'm using AAPS 3.1.0.2 on Android 11. Lately I've noticed that the notes I put in AAPS under the carbs button don't transfer to [#1279](https://deref-web.de/mail/client/nTfs0wpd6U0/dereferrer/?redirectUrl=https%3A%2F%2Fgithub.com%2Fnightscout%2FAndroidAPS%2Fissues%2F1279). In the treatments, user input they are listed. But they don't arrive in Nightscout. I hope you can help me? Many thanks for your help and support. Greetings from Joerg Fulge [image] [image]

jfulge, 2023-01-16:
> What does your queue states on the Nightscout tab within AndroidAPS? Are there any treatments in queue?
> 
> Which Nighscout version are you using?

Good morning, I'm using the nightscout version, 14.2.6

jfulge, 2023-01-16:
Good morning, Martin. I'm using the Nightscout version, 14.2.6.Thank you for your help and support. Best regards from Jörg Fulge

jfulge, 2023-01-16:
I just built AAPS 3.1.0.3 and installed it on my phone. Again, the notes I enter about the carbohydrates are not sent to Nightscout.

maheini, 2023-01-16:
Hi @jfulge 
I just tested the behaviour with the beta 2022.12.15 and Nightscout version 14.2.6.
Everything is working as intended - notes are visible on Nightscout - a bit hidden thought. Comments are only visible if you **hover over the entries**. 
Have a look here while hovering:
[image]

Does this resolve your issue? Feel free to close or write an answer. If nightscout isn't showing your notes properly, you might want to open a issue on https://github.com/nightscout/cgm-remote-monitor/issues

jfulge, 2023-01-17:
Hi Martin, Thank you for your effort, help and support. At Nightscout I try it out. I miss the notes in the report of Nightscout. Everything is in the report. Only the notes, what I enter, when I ate what. These are missing. If I have Nightscout open. Then go to the Treatments tab in the reports, because the entries I make in the notes, what I have eaten, are missing. I now have version 3.1.0.3 of AAPS. It doesn't work there either. Until AAPS 3.0 it worked. Another looper uses the version of AAPS, 3.1.0.3-dev-f, with which it works for him. I also can't imagine that it's the phone. I am using a Xiaomi Redmi Note 9 Pro. With Android 11.I have no idea what it could be and what I can do to make it work. In any case, it went to AAPS 3.0I've already been in contact with the developer of the Nightscout Reporter.  He says that he cannot access this data on Nightscout.  Perhaps Nightscout is also at fault?  Thank you for the help and support.Best regards from Jörg Fulge --Jörg FulgeMail: ***@***.***

jfulge, 2023-01-17:
Hello Martin, I just tried Nightscout with my cell phone.  If I go to such a marker in Nightscout, the input data is displayed.  Except for the note what I ate.  Here is my link to Nightscout.  https://jfulge.ns.10be.de:23228/Thanks for the help and support. --Jörg FulgeMail: ***@***.***

jfulge, 2023-01-18:
Hi Martin, I now have version 15.0.0 running in Nightscout. In AAPS I use version 3.1.0.3. There are still no notes displayed in Nightscout. Thank you for your efforts. Best regards from Jörg Fulge --Jörg FulgeMail: ***@***.***

jfulge, 2023-01-20:
Hello Martin Thank you for your help I will now close the problem. Best regards from Jörg Fulge --Jörg FulgeMail: ***@***.***

maheini, 2023-01-20:
You are welcome. Unfortunately, I had no dev environment to test 3.1.0.3 quickly. But I do think this issue is solved within the current dev version. Version 3.1.0.3 isn't improved anyways... just build dev version or new master after it's release.

Have a great day 😄

jfulge, 2023-01-20:
Hello Martin, Thank you for your quick reply. I love building the developer version. Can you please give me the link? I'm definitely trying. Thank you for your help and support. Best regards from Jörg Fulge --Jörg FulgeMail: ***@***.***

maheini, 2023-01-20:
To build the dev version, you just need to click in the bottom right corner within Android Studio and select origin/dev or dev. Be aware that this version might include less/untested code.
[image]

closed 2023-01-21 by MilosKozak