nightscout/AndroidAPS #2350: Dev loosing Accu_Chek Combo pump pairing after phone restart

tkvasnicka, 2023-01-16:
on my new Pixel 6 (with android 13)
i was not able to pair my pump (Accu_Chek Combo) with Ruffy 
i moved to the DEV version and paired with V2 driver 
pairing is relatively smooth - but it always works only with a second try
however with every phone restart so far AAPS is loosing the pairing, showing the pump as unpaired.
the pump stay listed among the bluetooth known devices though. 
pairing again works only after removing the pump from the list.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

swissalpine, 2023-01-16:
Can't confirm this problem on my phones:
Unihertz Jelly 2 and Samsung Flip 4.
Restarts doesn't change the pairing.

parapenT1sta, 2023-01-17:
Using latest DEV, Combo on a Google Pixel 4a with no issues at all.

MilosKozak, 2023-01-21:
Closing as not replicable

closed 2023-01-21 by MilosKozak