nightscout/AndroidAPS #2351: [DEV][NSC V3] Pod/Canula change not reported to NS

jbr7rr, 2023-01-16:
Build 4fbb0128e6-2023.01.15 merged at 378002df336294fe31495c50002ef139e806e2a4
Pumps used: Insight and DASH
Time: Switched from Insight to DASH at:  16-01-2023 11:47 and placed a new pod
Logs: [AndroidAPS._2023-01-16_JB.zip](https://github.com/nightscout/AndroidAPS/files/10426348/AndroidAPS._2023-01-16_JB.zip)
NS Version: 14.2.6

Canula and pod changes are not reported to NS. When changing the Canula of the Insight pump few days ago I did not see a canula change in NS. Today I switched from Insight to Dash, And the pod change is not registered in NS. 
CAGE 4+ days from when I used NSC V1

MilosKozak, 2023-01-21:
I tested it and it works . it appears in NS and in AAPS client too

jbr7rr, 2023-01-21:
Well that's strange, 2 Days ago I switched to EO, that switch/change was also not noted. Any settings I can check on my side (NS or AAPS?) I looked trough the settings and did not find any option for that though..

MilosKozak, 2023-01-21:
try to enter it from action tab . you can delete it afterwards

jbr7rr, 2023-01-21:
Unfortunately manually entering in the actions tab does not work as well.. 
Today I did change to insight at: 20:48 and manual via actions at: 21:04
[AndroidAPS_LOG_1674331834250.log.zip](https://github.com/nightscout/AndroidAPS/files/10472972/AndroidAPS_LOG_1674331834250.log.zip)

Do you want my settings file via PM?

MilosKozak, 2023-01-21:
to on default logging. it's quite impossible to search for something

MilosKozak, 2023-01-21:
there is no treatment created at 21:04

MilosKozak, 2023-01-21:
does the cannula age change localy?
do you see the record sent to NS within a few minutes?

jbr7rr, 2023-01-22:
Oops, yeah I can see why it's hard to read, I resetted the log settings this morning and did a manual trigger at: 11:21
[AndroidAPS_2023-01-22_JB.zip](https://github.com/nightscout/AndroidAPS/files/10474164/AndroidAPS_2023-01-22_JB.zip)

Yes, I see the change locally (note the missing NS)
[image]

NS Log screen after canula change triggered:
[image]

MilosKozak, 2023-01-22:
I see:
```
11:18:26.818 [DefaultDispatcher-worker-4] E/NSCLIENT: [NSClientV3Plugin$dbOperationTreatments$1$1.invokeSuspend():682]: Upload exception
info.nightscout.sdk.exceptions.UnknownResponseNightscoutException: null
	at info.nightscout.sdk.NSAndroidClientImpl$createTreatment$2.invokeSuspend(NSAndroidClientImpl.kt:307)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
```
unfortunately Http response is not logged. could you look via logcat why is NS refusing record?

jbr7rr, 2023-01-22:
```
14:29:41.025  I  <-- 400 https://jbr7rr-ns.up.railway.app/api/v3/treatments (287ms)
14:29:41.025  I  x-dns-prefetch-control: off
14:29:41.025  I  expect-ct: max-age=0
14:29:41.025  I  strict-transport-security: max-age=31536000
14:29:41.026  I  x-download-options: noopen
14:29:41.026  I  x-content-type-options: nosniff
14:29:41.026  I  x-permitted-cross-domain-policies: none
14:29:41.026  I  referrer-policy: no-referrer
14:29:41.026  I  x-xss-protection: 0
14:29:41.026  I  access-control-allow-origin: *
14:29:41.026  I  access-control-allow-methods: GET,PUT,POST,DELETE,OPTIONS
14:29:41.026  I  access-control-allow-headers: Content-Type, Authorization, Content-Length, X-Requested-With
14:29:41.026  I  content-type: application/json; charset=utf-8
14:29:41.026  I  content-length: 70
14:29:41.026  I  vary: Accept-Encoding
14:29:41.026  I  date: Sun, 22 Jan 2023 13:29:40 GMT
14:29:41.026  I  x-envoy-upstream-service-time: 145
14:29:41.026  I  server: railway
14:29:41.027  I  {"status":400,"message":"Field date cannot be modified by the client"}
14:29:41.027  I  <-- END HTTP (70-byte body)
14:29:41.035  E  [DefaultDispatcher-worker-3]: [NSClientV3Plugin$dbOperationTreatments$1$1.invokeSuspend():689]: Upload exception
								 info.nightscout.sdk.exceptions.InvalidParameterNightscoutException: 
									 at info.nightscout.sdk.NSAndroidClientImpl$createTreatment$2.invokeSuspend(NSAndroidClientImpl.kt:323)
									 at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
									 at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
									 at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
									 at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
									 at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
									 at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
									 at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
									 at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
```

jbr7rr, 2023-01-22:
Im thinking it might be a problem on my server side. I use non standard settings. as V3  is using https will that be of influence?

My ENV:
```
ALARM_HIGH=off
ALARM_LOW=off
ALARM_TIMEAGO_URGENT=off
ALARM_TIMEAGO_URGENT_MINS=30
ALARM_TIMEAGO_WARN=off
ALARM_TIMEAGO_WARN_MINS=15
ALARM_TYPES=simple
ALARM_URGENT_HIGH=off
ALARM_URGENT_LOW=off
API_SECRET=xxxx
AUTH_DEFAULT_ROLES=denied
BASAL_RENDER=icicle
BG_HIGH=15
BG_LOW=3,9
BG_TARGET_BOTTOM=3.9
BG_TARGET_TOP=10
BOLUS_RENDER_OVER=1
BRIDGE_PASSWORD=
BRIDGE_SERVER=US
BRIDGE_USER_NAME=
CUSTOM_TITLE=>(^,^)<
DEVICESTATUS_ADVANCED=true
DISPLAY_UNITS=mmol/L
EDIT_MODE=off
ENABLE=rawbg food bwp cage sage iage iob cob basal ar2 pushover bgi pump openaps cors
LANGUAGE=nl
MMCONNECT_PASSWORD=
MMCONNECT_SERVER=US
MMCONNECT_USER_NAME=
MONGODB_URI=xxxx
NIGHT_MODE=off
PUMP_FIELDS=reservoir battery
SHOW_FORECAST=openaps
SHOW_PLUGINS=dbsize rawbg cage sage iage iob basal pushover bgi pump openaps
SHOW_RAWBG=always
THEME=colors
TIME_FORMAT=24
BAGE_URGENT=960
BAGE_WARN=672
CAGE_URGENT=120
CAGE_WARN=72
DBSIZE_URGENT_PERCENTAGE=95
DBSIZE_WARN_PERCENTAGE=90
IAGE_URGENT=72
IAGE_WARN=144
SAGE_URGENT=480
SAGE_WARN=240
UPBAT_URGENT=10
UPBAT_WARN=20
XDRIPJS_WARN_BAT_V=280
```

MilosKozak, 2023-01-22:
send larger chunk of log pls

jbr7rr, 2023-01-22:
Triggered site change at 18:40

Logcat:
[logcat.log](https://github.com/nightscout/AndroidAPS/files/10474652/logcat.log)

I also noted that the first note that was not logged in NS is not an actual site change but something else, Maybe that is holding up the queue? As there are also some number under "waiting" in the NS tab:

[image] 

[image]

MilosKozak, 2023-01-23:
I changed processing to continue with upload when error occurs.
Update

closed 2023-01-23 by jbr7rr