nightscout/AndroidAPS #2353: [dev-g] NSClient / NSClientV3 preferences

Philoul, 2023-01-17:
Minor issue concerning both NSClient and NSCLient V3 in the shared preference management:

I you open the preference from NSClient, you have the NS API Secret parameter, this parameter is replaced by NS Access Token if you open the preferences of NSClient v3.

But if you open one of the sub menus (synchronization for example) and then go back to main preferences, then you get both parameters (NS API Secret AND NS Access Token) at the same time (it's a bit confusing...)

[image]

vanelsberg, 2023-01-18:
Confirming...

databoy2k, 2023-01-18:
It might be linked to a bug in the Preferences menu that results in all of the preferences "categories" being expanded. 

To reproduce:
1. Go into your preferences menu - note that all categories are "minimized" or have down-arrows
2. Expand, e.g. "Overview" and note that none of the other categories expand.
3. Expand a category whose entries change depend on your configuration, e.g. "OpenAPS SMB" and then choose an entry that takes you to another menu (e.g. "Advanced Settings"
4. Press the "back" button on your phone or use the back gesture to return to Preferences.

You'll see that everything is now expanded, not just the category that you went into.

My guess is that the menus all have conditions in them that choose which entries are shown based on config, but those conditions aren't always checked when the menu is loaded a second time via a "back". That would explain both this bug and the fact that the menus don't stay collapsed.

robertrub, 2023-01-18:
Confirming

[image]

[image]

Philoul, 2023-01-18:
@databoy2k I don't think the group / ungroup preferences in 3 dot menu is the same issue (it could be a constraint or a parameter of standard library that manage the preferences...)
But when all preferences are ungrouped in 3 dot menu after navigation, the NSClient is also unfiltered and shows both parameters (NS API Secret and NS token).

Alamo04, 2023-01-19:
Also confirmed

MilosKozak, 2023-01-21:
https://github.com/nightscout/AndroidAPS/commit/f877bb5fb9492f42fc885133e5bb859ffdc0ffa1

closed 2023-01-21 by MilosKozak