nightscout/AndroidAPS #2354: For Discussion: SGV (glucose) Smoothing & AID

Tornado-Tim, 2023-01-18:
Hello everyone,

I am writing this here after seeing an overwhelming push for AAPS to have some form "CGM Smoothing" put in, because the Dexcom G7 tended towards having jumpy values.

I am going to highlight these PR's here (Some of which are closed and not merged, however some have already been merged into Dev): https://github.com/nightscout/AndroidAPS/pull/2141 and https://github.com/nightscout/AndroidAPS/pull/2268

I want to highlight here the remits of each connected device and associated piece of Software:

- CGM and CGM Collector (be it software or Hardware, Official or DIY) - Responsible for the collection of glucose from said transmitter - ascertain validity/sanity of said values coming from hardware, perform calibration if needed. Pass this data along to AID System

-AID System: Gather and collect broadcasted glucose data from CGM Collector, run dosing algorithm and then connect to Insulin Pump to enact the algorithm required changes within pump specifications.

In total there are 3 Components to this system (* to show remit of where the AAPS Project sits in this equation):
1. CGM (hardware,software)
2. Insulin Pump (hardware)
3. Algorithm/AID System*

With the changes that have been merged now into dev, this is now the result of the remit:
1. CGM (hardware,software)*
2. Insulin Pump (hardware)
3. Algorithm/AID System*

### Reason:
AAPS is now generating its own CGM Data that is not from the hardware manufacturer or designated authority for CGM Data.

### Recommended Guidelines:
Conforming to the recommended FDA Ruleset where there is a "blackbox" on connected devices, everything should be able to audited and traced as to why each AID System behaves as it does so if anything goes wrong, causation can be traced back easily to individual components that complete the AID schematic.

### What results with the above changes:
- CGM validity and sanity checking now reside within the AAPS Project

Therefor the scope and responsibility is much higher for this project than it ever was before.

As an AAPS Developer- I believe AAPS should stick within its remit and not do any CGM Smoothing or any CGM tampering whatsoever. 

Smoothed data is fake data -it has not been generated from the hardware or CGM source. If an adverse event now comes - AAPS can be attributed as causing this issue, where as before the smoothing merge into dev, it could clearly be stated, it was pump or CGM source.

### Resolution:
-Revert any CGM modification commits

### Background:
oref (the AID algorithm used in AAPS) - is designed to run off the un-smoothed data and can handle jumps, it can even run off raw. Smoothing defeats what oref already does. 

If people wish to have smoothing (as has apparently become needed in the G7), this should be done in said CGM direct or Application (DIY or official), not AID Application.

This issue has been raised for this projects safety and efficacy.

scottleibrand, 2023-01-18:
It appears that smoothing is serving two functions here:

- Make the graphs less ugly and easier for humans to see what was really going on.
- Change the inputs being used for Automatic Insulin Dosing (AID).

We probably want to treat the two functions differently.

I don't have any particular problem with smoothing being applied to displayed data, as long as the raw data is also displayed. Based on https://github.com/nightscout/AndroidAPS/pull/2268#issuecomment-1342552649 that appears to be the default behavior. If that is correct, I probably don't have a problem with the implementation there for that function.

I do share many of @Tornado-Tim's concerns about smoothing the input data to oref0 for purposes of insulin dosing. In particular, I am concerned that doing so may defeat some of the safety measures in oref0, such as the detection of unrealistic jumps in BG and disabling of SMB until BG flattens out or declines.

Has anyone done any head-to-head studies of dosing for smoothed vs. unsmoothed data to determine whether smoothing causes AAPS to dose more or less insulin, both in the normal case and in the case of pathologically noisy CGM data? If not, we should set up some retrospective studies using NS / OpenAPS Data Commons data. If we use the oref0-backtest simulator, we can compare three outcomes: the real historical outcome, the simulated outcome without smoothing, and the simulated outcome with the AAPS smoothing algorithm (which someone will need to implement in the simulator).

Der-Schubi, 2023-01-18:
> Has anyone done any head-to-head studies of dosing for smoothed vs. unsmoothed data to determine whether smoothing causes AAPS to dose more or less insulin, both in the normal case and in the case of pathologically noisy CGM data?

For me it really helps to prevent hypos from sudden jumps in G6 data.
For example, I often have BG values like this:
120, 110, 104, 85, 110, 95, 93, [...]
85 and 110 are both clearly wrong here, but this jump causes a SMB of roughly 1.5U, resulting in a drop afterwards.
The new smoothing algorithm can prevent this easily.
As xDrip has no ability to filter such jumps, I'm glad that AAPS now has.

To be honest, I agree with you that this should actually be implemented in xDrip. But sad to say, when I look at the handling of pull requests in xDrip and the endless discussions in them, it would probably take us years to merge something like this into xDrip.

MilosKozak, 2023-01-20:
* I agree with your description
* I agree this should/will not be part of commertial system (if this happen) just because the source CGM should be trusted unconditionally

But the reality is:
* we are already recalculating data to be 5 min because algorithm expects it (even with G6)
* people are using different BG sources not matching these criteria and smoothing of data improves algorithm performace (COB calculation because of more adequate deviations, predictions because of not extreme deltas etc.)
* It can be easy disabled for commertial system (only force "No smoothing" plugin and hide control)

Tornado-Tim, 2023-01-20:
But the CGM source/app is responsible for it (if or any smoothing), I am not sure why AAPS is now taking on this responsibility when said smoothing commits could have been put towards Open Source CGM Application.

MilosKozak, 2023-01-20:
opensource is only xDrip which can handle smoothing on some BG sources. but there are many others

MilosKozak, 2023-01-20:
I was thinking about disabling this option for "good" sources like G6 but at the people told they want to use it for G6 as well at the end of life G6 sensor when it become noisy

Tornado-Tim, 2023-01-20:
But if you fake a end of life CGM sensor by smoothing it, people will have no idea when the sensor has actually gone bad. This is opening up so many holes for this project.

Der-Schubi, 2023-01-20:
> But if you fake a end of life CGM sensor by smoothing it, people will have no idea when the sensor has actually gone bad.

No, the original source is doing all the display work, the usercan see the real values. The value used and shown in AAPS plays a minor role here.

Tornado-Tim, 2023-01-20:
For example: you will now be enabling AID on a actually dysfunctional sensor. People will have a tendency to just ride with it because it's smooth. Despite being able to see the actually data is jumpy as, there is little to no incentive to change said sensor.

MilosKozak, 2023-01-20:
I'm sharing worries with @Tornado-Tim . We must find way to make things safe

Der-Schubi, 2023-01-20:
> I'm sharing worries with @Tornado-Tim . We must find way to make things safe

Then at least keep it in dev, until someone miraculously gets something like this merged into xDrip. :sweat_smile:

jbr7rr, 2023-01-21:
Merging it in xDrip will not solve the issue, there are many more BG sources..

I think AAPS already took responsibility in sanity checking the value's (see SafetyPlugin). AAPS is preprocessing raw value's for the algorithm, and now an extra preprocessing step is added, what makes that a part of CGM responsibility? To me smoothing is part of the algorithm. Smoothing can be mathematically verified. And like suggested here, we could even run it against an oref simulator and see what happens.

Also note that BYODA for G6 sent smoothed value's and that xDrip with G6 send raw, but in the G7 app smoothing is removed completely. There might be reasons for that. But with (slightly) jumpy value's the danger of overdosing on a fake jump is more present when no smoothing is done. 

I share the worry that users will keep using dying sensors and just use smoothing until the sensor really stops. The sensor may spit out crap for all we know.. However this is preventable with good documentation, education and by putting safe guards in place (see below). If the problem is the user interpretation, we could also flip the grey and green values, making the grey ones the smoothed and green the raw.. But well I think this is more confusing..

I think the core question is, Do we trust the data from CGM sources in AAPS?
If the answer is yes, we don't need to do anything (even don't smooth). But I think the answer is no. And we need safety checks and preprocessing for the algorithm, Including (optional) smoothing, for me smoothing is a safety feature..

I thought a lot about sensor safety lately, and I think the following improvements could be made in the safety check:
- Check if sensor errors where present (G6/G7 sensor errors) (if there are, stop loop and ask user for verification of BG value)
- Detect jumpy values on raw data, (Notify user on moderate jumps, and stop loop on severe jumps, and don't let the user re-enable it until the jumpy values are gone)

However this will greatly impact the user experience... And I think some users will get frustrated by this.. But I think it is needed. I got AAPS to dose a lot once after my sensor came back at 12mmol (actual was 5mmol) after a G6 sensor error.

blaqone, 2023-01-29:
I have previously used the g6 and now use the g7. Since version 1.10 of the g6 app there is no more smoothing. 
All commercial systems that want to use the g6 or the g7 in the future must include internal smoothing. 
Jumping values with the g7 are with every sensor! (99,80,110,125,105,…). If I would use these raw values in AAPs, I would have repeatedly too many and high SMB, which would eventually bring me into a dangerous range. 
The smoothing itself does not provide false-bad values that can become dangerous, but using the raw values is the problem. 
I use Tsunami Smoothing with the g7 and even with minute values of the FL2. Both tested over months. Not once did I have a "threat" from the smoothing.

khskekec, 2023-01-29:
Hey guys,

To be honest my knowledge about AAPS and all the technical stuff around is limited. As I integrated the broadcasting into G7 app and @blaqone started to test we realized quite fast that this jumpy values make the G7 nearly unusable.

This discussion here just ignores this case and taking edge-cases into account.

We are not talking about responsibility here because there is no party who would feel responsible or would help us to solve this problem beside ourselfs...

Therefore I think it is better to not try to answer this question at all but focus on solutions with safety and personal responsibility in mind.

From my point of view it is a bad idea to have smoothing as general option in AAPS. The availability should be connected to the cgm one is using. By this strict dependency we could rely on AAPS documentation resources and hint out advantages and disadvantages of smoothing and also clearly describe what it is able to do and its potential misuse.

We should not take personal responsibility from the users at all. They are already dealing with a dangerous system with big potential to be misused.

The question we have to answer is how to minimize this but transfering as much control as possible to knowledged and responsible people to get their desired hardware up and running tooo help them and provide as much information as possible about technologies they are going to use.

Kind regards

Selçuk

blaqone, 2023-01-29:
[image]

This is a normal day of a working g7 Sensor without smoothing.