nightscout/AndroidAPS #2355: [dev-g] BG Source gets reset to BYODA when switching to latest build

Der-Schubi, 2023-01-18:
Switching to latest dev build (d57ec150a5789a61f15c2781bf20ea5a90ae0644), the BG Source (in my case xDrip+) gets reset to BYODA.
Is it possible to retain that setting?

MilosKozak, 2023-01-20:
It didn't happen to me. is it repeatable?

parapenT1sta, 2023-01-20:
It happened to me, I had xDrip as source then was switch do BYODA

MilosKozak, 2023-01-20:
right. it's due to https://github.com/nightscout/AndroidAPS/commit/c71fc275a544bd3ae477549e8786e4188a69fe70 where XdripPlugin was renamed to XdripSourcePlugin to have original name preserved for xDrip synchronization plugin. 
I believe new naming convention is more logical but it means we cannot resolve this issue.
We must mention it in release notes

closed 2023-01-20 by MilosKozak