nightscout/AndroidAPS #2358: BUG NSv3 dors not do total synchronization

robertrub, 2023-01-18:
I had to try several times in order to do a total such if NS using NSv3

Then it started the synch but stopped in the 12th (today is 18th)...

[image]

[image]

Look at the logs between 22:14 and 22:16. I'm not sure if the sync/upload had started when I exported the logs.

[image]

MilosKozak, 2023-01-18:
it this is repeatable after cleaning db and resetting to full sync, pm me your config

robertrub, 2023-01-19:
Just tested. Cleared db for data over 3 months (minor bug : no message telling that it has been done) then tapped Total sync. Got the alert telling that it might take a long time. OKed. Nothing happened.

NS setup to not upload on battery but it is charging ATM.

Which config do you need ? The whole exported settings file or just the NS settings?

MilosKozak, 2023-01-20:
send me your config and pass

robertrub, 2023-01-20:
@MilosKozak  Done via DM in Discord.

closed 2023-01-20 by MilosKozak