nightscout/AndroidAPS #2359: NSClientV3: xDrip+ is not receiving data from AAPS (only BG values)

vanelsberg, 2023-01-18:
**AndroidAPS** 3.1.0.3-dev-g
**Build**: d57ec150a5
**Flavor**: Fullrelease
**xDrip+:** version 344deb4-2022-11-16

*General info:*
- Setup: AAPS/BYODA/G6
- AAPS is uploading BG & data to Nightscout v14.2.6.
- AAPS's NSClient and the "Send data to xDrip+" plugin enabled: This configuration makes xDrip+ to receive all data including basal rates and BG values. I use xDrip+ for uploading data to Tidepool: this works just fine. 

**Problem description:**
When I switch to the new NSClientV3 plugin xDrip+ is no longer receiving data from AAPS. It only is receiving BG values.
Switching back to NSClientV1 will have data backfilled to xDrip+.

vanelsberg, 2023-01-18:
Btw: I get the NSClientV3/xDrip plugins are still under development. Will retest as when possible

swissalpine, 2023-01-19:
Please check all sync settings (xDrip+ Plugin, NSClientV3). After the last update I have to reenable most of them. Especially the setting NSClientV3 > Advanced Settings > Enable local Broadcasts was crucial for the communication with xDrip+. Now everything works as before: Even with the latest dev version, xDrip+ receives all data!

MilosKozak, 2023-01-20:
this behavior was bound only to v1 nsclient.
https://github.com/NightscoutFoundation/xDrip/discussions/2636

closed 2023-01-20 by MilosKozak