nightscout/AndroidAPS #2361: Change from NSv3 to NSv1 : no more connection

robertrub, 2023-01-19:
Last dev_g. I was on NSv3 for a couple of days. Wanted to get back to NSv1 to stop having holes in NS web.

As a result, NS in AAPS can't connect to NS web. The web site is functional. I get the "usual" error (can't recall exactly, you'll see in the logs). The secret API is filled and correct.

If I switch back to NSv3, it connects normally.

See logs around 9:20

[image]

Alamo04, 2023-01-19:
NSC v3 never worked for me,  nothing was syncing with that version.
So I tested NSC v3 and going back to v1 immediately. 

NSC v1 is still working fine.

MilosKozak, 2023-01-19:
`D/NSCLIENT: [NSClientPlugin$onStart$9.accept():127]: ERROR io.socket.engineio.client.EngineIOException: xhr poll error`
means connectivity issue

robertrub, 2023-01-19:
Yes, I know but changing back to NSv3 brings the connectivity back. Changing to v1 and I get the error 

It made me think that as v1 is using the secret API and V3 is using the token, maybe one of them got stuck in a variable and v1 tries to use the token, instead of the secret API (this is a very wild guess though!)

robertrub, 2023-01-19:
@Alamo04  Wait 5 to 10 minutes and you'll see the uploading.

MilosKozak, 2023-01-19:
btw I tested NS dev and AAPS wont connect too. What's your NS version?

Alamo04, 2023-01-19:
@robertrub 
At my first try, I saw after 30 minutes that NS didn't get new BG, or any other data.
I entered API key and also create a new Token for AAPS and entered it also, next try and nothing changed.
Also tryed a restart of phone and AAPS.
V3 isn't working for me.

NSC v3 also says "not connected".

But you say I have to wait and so I tested it again. Waiting now 20 minutes, without sync to NS.
API key and token are entered and the NS site is working, so the client has to upload the data and bg, but nothing happen.
All settings are the same as in NSC v1 + Token.

Going back to v1 and everything works fine again.

Alamo04, 2023-01-19:
> btw I tested NS dev and AAPS won't connect too. What's your NS version?

15.0.0

robertrub, 2023-01-20:
I'm using latest_dev which is v15.0.0.

I have tested again and again with the same result. V3 works. V1 gives the connection error.

I change to V1, force close AAPS, clean cache, restart AAPS. Can't connect.

Change back to V3 and even without restart, the connection is good and data starts flowing.

@MilosKozak  You had asked for my settings but didn't say which one. Aaps or NS settings in AAPS?

robertrub, 2023-01-20:
I'm using latest_dev which is v15.0.0.

I have tested again and again with the same result. V3 works. V1 gives the connection error.

I change to V1, force close AAPS, clean cache, restart AAPS. Can't connect.

Change back to V3 and even without restart, the connection is good and data starts flowing.

@MilosKozak  You had asked for my settings but didn't say which one. Aaps or NS settings in AAPS?

EDIT : Milos, as your message on Discord AAPS, it seems normal that v1 doesn't work with NS v15

[image]

Closing this issue.

closed 2023-01-20 by robertrub