nightscout/AndroidAPS #2369: PasswordReset does not work for Accu-Check Combo

sdpeixoto, 2023-01-20:
I tried to reset the master password on version 3.0.0.1 (Build: 479bfc410-2022.03.03-12:36) following the instructions given in the docs for v3.0 and above. I created the "PasswordReset" empty file and placed it in AAPS/extra folder from mobile phone and restarted the app. The file disappeared and the app showed a text message stating that the password had been set. But the password still isn't the Combo serial number.

I there something I'm doing wrong or overlooking? Or is there a different password for the Accu-Check Combo pump?
Help would be really appreciated.

MilosKozak, 2023-01-20:
AAPS 3.0 is unsupported.
Upgrade to 3.1 first

closed 2023-01-20 by MilosKozak

sdpeixoto, 2023-01-28:
Hi @MilosKozak,

Thanks for the feedback.

Following your suggestion, I have upgraded APPS to 3.1.0.3 version but master key reset functionality still doesn't work as described in the documentation.
I copied a "PasswordReset" empty file to AAPS/extra folder, restarted APPS application, file got droped, saw a popup message saying that password got changed, but then I when enter insulin pump serial number, it says that password is not correct.

Help would be really appreciated.

sdpeixoto, 2023-01-28:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10528082/AndroidAPS.log)
[AndroidAPS._2023-01-28_13-30-08_.0.zip](https://github.com/nightscout/AndroidAPS/files/10528085/AndroidAPS._2023-01-28_13-30-08_.0.zip)

In attachment you can find AAPS log files.

sdpeixoto, 2023-02-01:
Any news please?

parapenT1sta, 2023-02-01:
@sdpeixoto Which Combo driver are you using? V1 or V2? With V1 and Ruffy the pump serial number was the Mac Address.