nightscout/AndroidAPS #2370: Bug ?? Connected to power but NS not uploading

robertrub, 2023-01-21:
I'm not sure if it is a bug or an Android limitation...

NS set to upload _when charging_ only. The phone is connected to power but the battery is full => NS doesn't upload. 

Technically, the phone isn't charging (as the battery is full) but it is still connected to external power.

It would be logical to still upload in this case as the "limitation" is there to limit battery drain and with power connected, there is no drain on battery.

MilosKozak, 2023-01-21:
[image]
and uploading

MilosKozak, 2023-01-21:
i did small update. connection status is updating faster now

robertrub, 2023-01-21:
Will test your update this afternoon. 

My phone does not upload when battery is charged (I can see when it is not uploading with the holes in the NS chart)

[image]

MilosKozak, 2023-01-24:
isn't this issue the one with NS?

closed 2023-02-13 by MilosKozak