nightscout/AndroidAPS #2371: [AAPS 3.1.0.3-dev-g Build 0569d2ba0a Both AMA and SMB algorithms are activated at the same time.

fjpezuela, 2023-01-21:
Reporting bugs
--------------
I use the OpenAPS AMA algorithm and every time I compile a new dev version of AAPS and install it, the OpenAPS AMA and OpenAPS SMB algorithms are checked and the second one is left running. This also tends to happen without updating, every so often without any apparent cause.

The problem occurred after updating AAPS, around 13:16.

AAPS 3.1.0.3-dev-g Build 0569d2ba0a

[AndroidAPS._2023-01-21_01-00-40_.9.zip](https://github.com/nightscout/AndroidAPS/files/10472145/AndroidAPS._2023-01-21_01-00-40_.9.zip)
[Record_2023-01-21-13-09-44.zip](https://github.com/nightscout/AndroidAPS/files/10472158/Record_2023-01-21-13-09-44.zip)
[AndroidAPS.zip](https://github.com/nightscout/AndroidAPS/files/10472160/AndroidAPS.zip)

MilosKozak, 2023-01-21:
how do you do upgrade? remove AAPS, install new and load config?

fjpezuela, 2023-01-21:
It happens to me all the time. A few days ago I removed AAPS, installed it from scratch clean and it also starts with the two selected algorithms.

MilosKozak, 2023-01-21:
i see 13:10 was loaded configuration with Insight enabled not APS selected which means it was defaulted to SMB

MilosKozak, 2023-01-21:
let me guess. configuration was saved with some experimental branch like AIMI. If i'm right if you now restore configuration it will be back to SMB

fjpezuela, 2023-01-21:
Yes, I used AIMI several months ago, but before making the switch to the dev version, I have been using EatingNow.

I tried installing the dev on top of it and had the problem. I tried saving the dev configuration and going back to the master and the same thing happened. I switched to dev from master and the same thing happened. I left AMA and saved configuration, cleared the AAPS cache and uninstalled. Then I installed the dev version again and the same thing happened. I don't know if I might have something cached somehow that is causing this problem. I'll try it tomorrow on another phone.

MilosKozak, 2023-01-23:
try dev wiith this commit https://github.com/nightscout/AndroidAPS/commit/c4353e2e3eb698393aeba587e9a19ab4304ed31e

fjpezuela, 2023-01-23:
I have just tested it and it is working fine. It has left me marked AMA.

The only thing, in case you want to take it into account, is that the OpenAPS AMA check, I don't have it checked (I have it in the hamburger menu) and now it has moved it to the tabs. As I said, no problem with that.

Thank you very much for all the support.

closed 2023-01-23 by MilosKozak