nightscout/AndroidAPS #2382: [3.1.0.3-dev-g] [ComboV2] User Entry Log

markvader, 2023-01-31:
ComboV2 driver automatically recognises insulin cartridge change && (if newLevel > currentLevel)

Entry automatically made in the Careportal log and User entry log recording the change.

**The Careportal log entry shows** 
1. the time "hh:mm"
2. the title "Insulin Cartridge Change"
3. a note that the "Insulin reservoir change inserted automatically by combov2 driver".

All okay here

**The User Entry log shows** 
1. the time "hh:mm:ss"
2. the title "CAREPORTAL"
3. The combo icon
4. the date and time MM-DD-YY HH:MI:SS and the title "Insulin Cartridge Change"
5. the note "Insulin reservoir change inserted automatically by combov2 driver" is right aligned to the same width as the combo icon

**Issues:** 
1. the date time duplication.  In the entry I have their is a 1 second difference between the two entries, is this why it appears twice.  Does their need to be an increase in tolerance for timedelta between phone and pump?
2. The title on 3.1.0.3 for combov1 is 'Reservoir Change' rather than CAREPORTAL
3. Will removing the date time duplication give enough space to the note.

I can see that pumps like the DanaRS to not use the note parameter in its pumpSync.insertTherapyEventIfNewWithTimestamp record, however I see it's value in careportal, particularly as users switch over from old combo to new.

[image]

@dv1

dv1, 2023-01-31:
> The title on 3.1.0.3 for combov1 is 'Reservoir Change' rather than CAREPORTAL

I am confused. Where does the combov1 (I assume you mean the old Ruffy based driver) use this? It does not enter any treatment.

markvader, 2023-01-31:
agreed, the ruffy driver is all user entry. 
[image]

I got someone to send me the danaRS screens, which has automatic cartridge change recognition (time: 22.21).  This user has also manually recorded the change (time: 22.24) (hence two different entries)
[image]

dv1, 2023-02-25:
Can you try to cherry-pick https://github.com/dv1/AndroidAPS/commit/1ff34633d07b4a8205d7591b9bae142884c19319 and see if this fixes the issue for you?

markvader, 2023-02-28:
[image]

The user entry for the automatic cartridge change remains the same with the long message squeezed to the right side of the screen.

dv1, 2023-03-29:
@markvader Can you confirm that the issue is fixed now in latest dev?

closed 2023-04-05 by MilosKozak