nightscout/AndroidAPS #2385: [3.1.0.3-g-dev]

swissalpine, 2023-02-02:
Treatments from Nightscout are not taken over

Temporary targets and carbohydrates entered in Nightsout (v15.0) are not realised in AAPS.

AAPS version: dev from 30.01.30 including NS 15.0.0 support
57e7bdd2fd-2023.01.30

Logs are send by E-Mail

[image]

[image]

MilosKozak, 2023-02-02:
this issue is only when you enter carbs through NS?

swissalpine, 2023-02-02:
Same with temp targets.

swissalpine, 2023-02-02:
[AndroidAPS._2023-02-02_01-00-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/10571407/AndroidAPS._2023-02-02_01-00-00_.1.zip)

MilosKozak, 2023-02-11:
works in ns15 branch

swissalpine, 2023-02-11:
Hm. I have merged the ns15 branch ...

swissalpine, 2023-02-15:
Also in the dev from yesterday (you have merged the ns15 branch) the carbs and temp target take over doesn't work for me.
Should I open a new issue?
Do you need new logs?

MilosKozak, 2023-02-15:
do you have websockets off?

MilosKozak, 2023-02-15:
https://github.com/openaps/AndroidAPSdocs/blob/prepare3.2.0/docs/EN/Installing-AndroidAPS/Releasenotes.md

swissalpine, 2023-02-15:
I didn't know the release notes yet. That explains why it doesn't work with websockets. Thanks.

But with websockets disabled, it doesn't synchronise for me either. The last Nightscout entries are only there after an AAPS restart. Nightscout plugin V1 works without problems.
I have just updated to the latest Nightscout dev without success.
I noticed a typo in treatmentnotify.js in the last commit:

      if (!eventType && lastTreatment.insulin) eventType = "Correcton Bolus";

That should be "Correction Bolus", right?
But that won't be the problem. I only enter targets and carbs via Nightscout.

parapenT1sta, 2023-02-15:
@swissalpine I had some issues when migrated to V3 but was a mistake from my side, I didn't had setup the token on NS with "admin" role. Since I change all the issues were solved with V3.

MilosKozak, 2023-02-15:
you have to wait a few minutes to let treatments appear as there is no permanent connection. did you?

swissalpine, 2023-02-15:
> @swissalpine I had some issues when migrated to V3 but was a mistake from my side, I didn't had setup the token on NS with "admin" role. Since I change all the issues were solved with V3.

Thanks, but that's not my problem. Upload is working fine, an admin token is set.

> you have to wait a few minutes to let treatments appear as there is no permanent connection. did you?

Yes. I can wait an hour, without any success, only after a restart of AAPS carbs and temp targets are synchronized. Restarting the Nightscout-Client changes nothing.

swissalpine, 2023-02-15:
Send by discord.

swissalpine, 2023-02-20:
Fixed by https://github.com/nightscout/AndroidAPS/commit/853c08e5ef061a625eeb37f4edc24711bccd9676
Great!

swissalpine, 2023-11-13:
Since some days I see this problem again: Carbs entered in Nightscout aren't synced (websockets disabled), only a full sync imports them (v3.2.0.2).
Is this a regression, it has worked for months.

MilosKozak, 2023-11-14:
in v3 entering data through NS UI doesn't work. see release notes

swissalpine, 2023-11-14:
But it has worked in Dev just one or two weeks ago. You have fixed it once with v3 dev (see commit above).

swissalpine, 2023-11-14:
And: Websockets are disabled. The release notes addresses this issue with websockets enabled.

OpossumGit, 2025-01-01:
I had this issue in the past and decided to readdress now that I updated to 3.3.0. It still exists so I’d like to bump this historic issue. 

AAPS not a receiving temporary target from NightScout when nsclientv3 is used. They simply are not received while nightscout shows them. 
“Receive temporary target” is set in preference. 
It works with NSclientv1.
AAPS 3.3.0, NS 15.0.2
Web sockets on or off - no difference.

MilosKozak, 2025-01-01:
provide logs from both apps, time, actions done etc otherwise you comment is wasting of your time