nightscout/AndroidAPS #2393: Backup failed

Viewquest, 2023-02-09:
Yesterday I got the new AAPS 3.1.0.3-dev-g
Installed Build: 1da267f05c-2023.02.08 and found that AAPS crashes and reboots on maintenance/export setup. Reinstalled the previous version and the error was gone.
I use the Eopatch, XDrip and Pixel 6a.
[image]

mjhuisman, 2023-02-09:
Same here. I cant export settings into json. Aaps doesnt have the rights to write an export file, but it has.

Viewquest, 2023-02-09:
The point is that I can no longer export settings. When I want to do that with this version, AAPS crashes and restarts.

Viewquest, 2023-02-09:
Now I installed the latest version (09_02_2023_d7fd4d0_app-full-release.apk) and the same problem. AAPS restarts when backing up.

MilosKozak, 2023-02-09:
reverted

closed 2023-02-09 by MilosKozak