nightscout/AndroidAPS #2404: Feature request: Allow creating notification/warning when Canula (or any other thing) has been in use for certain amount of time

andyrozman, 2023-02-12:
Lately I have been using less insulin than before, so It takes me about 5-6 days to use up all 290 (or so) units of insulin from my Combo. As you know Canula should be changed every 3 days or so, so I would like to have warning that I need to change Canula.

So what I would like is to have some time warning mechanism that can warn you to change Canula/Sensor/Battery or something.

This could be either:
1. Own functionality
2. Automation rule

Action executed could be many things:
1. Alarm
2. Alarm with dialog
3. Notification
4. Posibility of repeating notification
5. All of above (configurable)

MilosKozak, 2023-02-13:
you can setup status lights. I'd keep notifications for most important events

jbr7rr, 2023-02-15:
I think canula age as an automation trigger could have many more uses then just a warning. For example if one gets more insulin resistant at day 2 of canula age, they could add a profile % rule. 

As automations are already a thing for advanced users I think going this route would be preferable when comparing to an alarm, notifcation or something configurable.

szantos, 2023-02-28:
+1 for new automation triggers (and/or actions)

olorinmaia, 2024-03-18:
@MilosKozak you can close this issue due to https://github.com/nightscout/AndroidAPS/pull/3267

closed 2024-03-18 by MilosKozak