nightscout/AndroidAPS #2408: no connection to eopatch

Viewquest, 2023-02-14:
Today I installed the new Dev-H version 14_02_2023-388ee3d_app-full-release.apk and couldn't deliver a bolus afterwards because AAPS couldn't get a connection to Eopatch. I reinstalled the old version and it works again.

Glimmerglykose, 2023-02-14:
> Today I installed the new Dev-H version 14_02_2023-388ee3d_app-full-release.apk and couldn't deliver a bolus afterwards because AAPS couldn't get a connection to Eopatch. I reinstalled the old version and it works again.

I will be updating tomorrow and testing if this is the case for me as well. I will let you know.

MilosKozak, 2023-02-14:
I tested activvation and bolus without any issue

Viewquest, 2023-02-15:
[image]
Here is a screenshot. Since yesterday I have connected the Galaxy again. Can this cause a problem?

Glimmerglykose, 2023-02-15:
> [image] Here is a screenshot. Since yesterday I have connected the Galaxy again. Can this cause a problem?

I'm not sure what you actually are displaying, where you think the issue could be? 
However, sounding like a total newbie, what menu option is it I'm looking at here? - I haven't seen anything like it in my AAPS and would love the overview it gives :P

Viewquest, 2023-02-15:
Why do you have to insult someone if you don't understand them. What high horse are you sitting on? I'm actually not a pro when it comes to technology. I described my problem and got a corresponding answer. It was important to me again to. show that my pump is no longer addressed after the update. If you look at the screenshot, you could see that. But I'll build the new available APK again and then I'll report if there's still an error. Also as a "beginner".

Glimmerglykose, 2023-02-15:
> Why do you have to insult someone if you don't understand them. What high horse are you sitting on? I'm actually not a pro when it comes to technology. I described my problem and got a corresponding [...]

I'm thoroughly confused now... 
I stated that I was a otal newbie because I didn't quite understand what you had posted a screeshot of? - if you read it as me calling you something, I am truly sorry - that was never my intention. I honestly just would love to know what menu it is that you posted a pic off, since I can't find it in my aaps and love the layout! No sarcasm of irony - although I understand that it can be hard to actually see that through text..

Viewquest, 2023-02-15:
Sorry, I totally misunderstood you then. You can see how my screenshot came about by going to the "Treatments" menu and then pressing the right button. After today's update, I currently have no more errors.

closed 2023-02-16 by MilosKozak