nightscout/AndroidAPS #2409: [Latest DEV] No Pump and OpenAPS info on NS15 with NSclientV3

parapenT1sta, 2023-02-15:
First I have to tell I have xDrip sending BG values to NS since AAPS doesn't send it even if it is selected on NS options.

Is anything I can do yo fix it?

No Pump and OpenAPS is sent to NS with NSClient V3 and the predictive lines are not in the latest BG reading dot.
Compare the behavior on NSClient V3:
[image]

On NSClient V1:
[image]

MilosKozak, 2023-02-15:
1) xdrip can send to NS itself thus this option is not available in AAPS
2) status is uploading well for me. more info needed
[image]

parapenT1sta, 2023-02-15:
@MilosKozak 
1. I was talking about xDrip since on V3, there is the info that all uploaders should use V3 and I think xDrip is V1, that was my confusion.
[image]

2. What more info do you need from me?

MilosKozak, 2023-02-15:
1) sending to xdrip is no more related to comm with NS
2) logs, screenshots from nscv3 tab

parapenT1sta, 2023-02-15:
Logs and screenshots: [AAPS_NS15.zip](https://github.com/nightscout/AndroidAPS/files/10741905/AAPS_NS15.zip)

parapenT1sta, 2023-02-15:
One more screenshot with V3, for now I am going back to V1:
[image]

MilosKozak, 2023-02-15:
it's an issue with your NS, not AAPS. You may try new deploy

MilosKozak, 2023-02-15:
or token is bad?

closed 2023-02-15 by MilosKozak

parapenT1sta, 2023-02-15:
I rechecked the token but I'm going to create a new one. I had redeploy recently.

Will give a feedback

parapenT1sta, 2023-02-15:
@MilosKozak Thank you for the tip about token, I just realized since I migrated to a new server I didn't set the token for admin role, issue now solved!! :-D