nightscout/AndroidAPS #2411: [DEV][Tidepool] Manual BG entries messing up data in tidepool

jbr7rr, 2023-02-15:
Build 4fbb0128e6-2023.01.15 merged at /
Pumps used: Insight, Dash and Virtual
Logs: I had logging disabled at the time of enabling the Tidepool plugin. I will add logs of a manual entry upload as soon as I have logs for it (have to wait 3 hours before tidepool uploads)

When a manual BG entry is present (these are coming from xDrip in my case) in the data during Tidepool does not handle this correctly.
Example 1: Manual entry is added at 8:20 @ 3.7mmol, Tidepool counts this as a whole day below target range in the stats
[image]

Example 2: Manual entries are added at: 12:30, 14:20 and 14:33 all in range, 3 days of in range is added to stats
[image]

If more information is needed please let me know

jbr7rr, 2023-02-16:
New entry gave problems as well:
Time of manual BG entry: 15-02-2023 19:47
[AndroidAPS_LOG_1676508664449.log.zip](https://github.com/nightscout/AndroidAPS/files/10755211/AndroidAPS_LOG_1676508664449.log.zip)

MilosKozak, 2023-02-16:
I checked Tideppol documentation and data is uploaded as requested
http://developer.tidepool.org/data-model/device-data/types/cbg.html
report issue to https://public-chat.tidepool.org/

gniezen, 2023-03-06:
@jbr7rr Are these manual entries from a BG meter? Then AndroidAPS is uploading them as the wrong record type. They should be handled as self-monitored BG values, not CGM values: https://developer.tidepool.org/data-model/device-data/types/smbg.html .

emmatovar27, 2023-03-26:
The correct type should be "smgb"

Currently, the value "cgb" in line 24 in this file

plugins/sync/src/main/java/info/nightscout/plugins/sync/tidepool/elements/BloodGlucoseElement.kt

The glucose reading looks like dots in the graph. 

[image]

jbr7rr, 2023-03-27:
@emmatovar27  Nice, will you create PR for this? Otherwise I can change and test, and create PR afterwards?