nightscout/AndroidAPS #2420: "Pump busy" until AAPS restart

tanja3981, 2023-02-18:
Reporting bugs
--------------
- Loop reverted to default basal for hours until I noticed (because of an UAM and high BG). A restart of AAPS immeditately fixed the issue.
- Log indicates the "Pump is busy", as if the last command is hanging and canceled on AAPS restart. 
- In the log this begins at 21:17 and stays like that until AAPS is restarted.
- This happends frequently (2x this week).

AAPS version: 3.1.0.3-dev-h, but also happened with dev-g
Pump: DanaRS v3, Model: 05, Protokoll: 12, Code: [11]
Nightscout: dev

MilosKozak, 2023-02-18:
```
21:10:38.986 [Thread-14481] D/PUMPBTCOMM: [BLEComm.sendMessage():779]: >>>>> BASAL__APS_SET_TEMPORARY_BASAL B2 C1  B4 00 96 
21:10:41.518 [RxCachedThreadScheduler-64] D/AUTOMATION: [AutomationPlugin$onStart$11.accept():176]: Grabbed new BT event: EventBTChange[deviceAddress=E5:DC:4C:F5:84:EE,deviceName=Versa 2,state=CONNECT]
21:10:44.040 [Thread-14481] W/PUMPBTCOMM: [BLEComm.sendMessage():839]: Reply not received BASAL__APS_SET_TEMPORARY_BASAL
21:10:45.041 [Thread-14481] D/PUMPCOMM: [DanaRSPacketAPSHistoryEvents.<init>():44]: Loading event history from: 17.02.23 21:04
21:10:45.042 [Thread-14481] D/PUMPCOMM: [DanaRSService.loadEvents():266]: Loading event history from: 17.02.23 21:04
21:10:45.043 [Thread-14481] D/PUMPBTCOMM: [BLEComm.sendMessage():779]: >>>>> APS_HISTORY_EVENTS B2 C2  17 02 11 15  04 23 
21:10:50.097 [Thread-14481] W/PUMPBTCOMM: [BLEComm.sendMessage():839]: Reply not received APS_HISTORY_EVENTS
```

Pump stops responding. From logs I'd say connection of Versa 2 is the reason
Best would be resolve it on Android side but as a workaround which could help AAPS recover from this situation you may try add following code:

BLEComm:841 
`danaRSPlugin.disconnect("Reply not received")`

let me know

tanja3981, 2023-02-18:
Thanks a lot, I will try this out and report back!

MilosKozak, 2023-02-26:
https://github.com/nightscout/AndroidAPS/commit/1fa2cb937095647843ef9febdc344b26fb6133a9

closed 2023-02-26 by MilosKozak