nightscout/AndroidAPS #2422: AAPS not syncing properly with AAPSClient

bchris21, 2023-02-18:
Versions: 
AAPS Dev-h 0ae21b7a7d
NSClient Dev-h 21b3a70ad9

AAPS values are different than AAPSClient even though they seem to properly sync.

Carbs and Temp Targets sent via NS treatment menu do not pass to AAPS.

[image]

[image]

[image]

[image]

[image]

[image]

NS hosted on 10be.de

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2023-02-18:
read realease notes , especially the part regarding websockets

bchris21, 2023-02-18:
> websockets

Is there a way to sacrifice battery life to faster sync or it's a fixed functionality with v3 to have less frequent communication?
I am monitoring my son at school and I would like to have values syncing as soon as possible.
Thank you

bchris21, 2023-02-18:
> > websockets
> 
> Is there a way to sacrifice battery life to faster sync or it's a fixed functionality with v3 to have less frequent communication?
> I am monitoring my son at school and I would like to have values syncing as soon as possible.
> Thank you

Websockets are on, on both AAPS and NSClient but still there are delays.

bchris21, 2023-02-19:
With websockets ON or OFF there is an inconsistency in IOB and Carbs between AAPS and NSClient.

[image]

[image]

Is there something I can do to troubleshoot? Shall I delete the database on AAPSClient ?

Thank you

MilosKozak, 2023-02-21:
I see different time on both screenshots

bchris21, 2023-03-03:
Sorry Milos, wrong screenshots were chosen. Titles were truncated when chosen via mobile app and attached the wrong ones.

I still have an issue though with wrong carbs added via NSClient. They are added to AAPS (and NS) but they are not deleted when removed from NSClient Treatments.

That's a bit of an issue when kid is at school.

Test was made on 20:22 with 2gr test carbs entered from NSClient. They were deleted a few seconds away from treatments menu, but they are still in AAPS.

[image]

[image]

I don't know if logs help here. I attach from both.

AAPSClient2 3.1.0.3-dev-h
Build: da76506e01-2023.02.21

AAPS 3.1.0.3-dev-h
Build: da76506e01-2023.02.21

[image]

[image]

Final question:
When passing a Temp Target with NSClient, it syncs only when BG change arrives thus every 5th minute. Is this expected or should sync right away when NSClient sends a change ?

Thank you

MilosKozak, 2023-03-04:
https://github.com/nightscout/AndroidAPS/commit/d8235bab29db86a9abb8765c09009fc404a7756e

closed 2023-03-04 by MilosKozak