nightscout/AndroidAPS #2424: When syncing to xDrip, only BG and basals are uploaded.

vanelsberg, 2023-02-19:
**AndroidAPS** 3.1.0.3-dev-h
**Build**: `a608f55531`
**Flavor**: fullrelease / Engineering mode

**Problem description**:
When syncing to xDrip, treatments (Carbs & bolus) are missing. Only BG and basals are uploaded.

**Screenprints**
[image]
[image]

**Logfiles:**
[AndroidAPS_LOG_1676819528732.log.zip](https://github.com/nightscout/AndroidAPS/files/10777198/AndroidAPS_LOG_1676819528732.log.zip)

overfrenk, 2023-02-19:
Same for me!

parapenT1sta, 2023-02-19:
Also the same here. To get SMB on xDrip I need to setup it to download treatments from NS.

MilosKozak, 2023-02-20:
https://github.com/NightscoutFoundation/xDrip/discussions/2636

parapenT1sta, 2023-02-21:
I think some progress is beging made with the latest xDrip Nightly version: https://github.com/NightscoutFoundation/xDrip/commit/fcd8063a4149c59d01bc668e191eaa3c6cee5c3e

vanelsberg, 2023-03-23:
@MilosKozak Really missing the option of syncing _all_ data to xDrip+
Any outlook/roadmap on a possible fix?

MilosKozak, 2023-03-24:
it will start work when xdrip team implement it

parapenT1sta, 2023-06-17:
Issue fixed with xDrip latest nightly version: https://github.com/NightscoutFoundation/xDrip/releases/tag/2023.06.17

At least it's working for me. I think this ticket can now be closed.

vanelsberg, 2023-06-25:
> Issue fixed with xDrip latest nightly version: https://github.com/NightscoutFoundation/xDrip/releases/tag/2023.06.17
> At least it's working for me. I think this ticket can now be closed.

Nope. Nothing changed for me.

(Where was it fixed, what commit?)

parapenT1sta, 2023-06-25:
https://github.com/NightscoutFoundation/xDrip/commit/ec2f90f7ecb12a8305e4ed9166342fafcd065918

I have AAPS syncing all the info to NS and broadcasting the info to xDrip. I enabled "Enable Broadcast to xDrip+" and "Send status line to xDrip+" in AAPS.

parapenT1sta, 2023-06-25:
[image]

[image]

[image]

parapenT1sta, 2023-06-25:
[image]

parapenT1sta, 2023-06-25:
[image]

parapenT1sta, 2023-06-25:
> > Issue fixed with xDrip latest nightly version: https://github.com/NightscoutFoundation/xDrip/releases/tag/2023.06.17
> > At least it's working for me. I think this ticket can now be closed.
> 
> Partly. Now do see carbs and insulin treatments as data point being synced to NS. But no basal rates. NA is showing default basal rates from current profile on NS
> 
> Also COB and IOB reported on NS is wrong...?
> 
> > ....I think this ticket can now be closed.
> > So, nope, not yet...

The only issue I have is that BG values sent from AAPS aren't shown in Nighscout: https://github.com/nightscout/AndroidAPS/issues/2602 

Hopefully @MilosKozak can take a look.

vanelsberg, 2023-06-26:
> The only issue I have is that BG values sent from AAPS aren't shown in Nighscout: #2602

With the update, xDrip is now showing showing correctly. So that problem is fixed improvement.

However, for me xDrip+ uploading to NS is ok, except for current basal rate:
NS only shows default basal values from current profile.

closed 2023-08-17 by MilosKozak