nightscout/AndroidAPS #2426: Dev-h tdd calculation doesnt work in stats tab

RalfCGM, 2023-02-19:
Tdd calculation doesnt work anymore.

Maybe related. Also notice really slow refresh of main screen calculations. (takes regularly 15 Till 30 secs to calculate all graphs).

Tried reset button from stats, didnt work.
Tried cleaning database (deleted all old data) from maintenance, didnt work.

Already noticed that tdd calculation was slow sometimes, the last months, but they eventually did come through. Now they dont appear anymore.

Am using dana-i. Issue has nothing to do with the tdd stats directly from dana. That works fine still.

[Screenshot_20230219-231241](https://user-images.githubusercontent.com/15680598/219978741-5dd2d1f8-e891-4784-bb03-0057b795ba2b.png)

[AndroidAPS._2023-02-19_01-00-00_.86.zip](https://github.com/nightscout/AndroidAPS/files/10778111/AndroidAPS._2023-02-19_01-00-00_.86.zip)

MilosKozak, 2023-02-20:
log doesn't have time 23:12
I cannot reproduce this issue

RalfCGM, 2023-02-21:
Can i help investigating? Can i provide more info?

What do you expect the maximum wait time to be to calculate tdd? 5 minutes?

RalfCGM, 2023-02-21:
I noticed that My history browser (which was always really slow), doesnt work anymore at all. Does this point towards a database issue?
[image]

vanelsberg, 2023-02-24:
Did you try resetting your database?

MilosKozak, 2023-02-25:
I cannot reproduce this issue. Try to open the screen again and provide log from that time

RalfCGM, 2023-02-27:
Tried cleaning up database, not resetting, because of the hassle of getting started up after reset.

RalfCGM, 2023-02-27:
Attached logs.

What I did:
I force stopped aaps
Cleared cache of aaps
Started aaps
Waited Till app was fully running. (Main screen refreshed)
Loaded stats
Waited couple of minutes.
Created first log file
Waited an other couple of minutes
Created second log file.
[AndroidAPS_LOG_1677485893041.log.zip](https://github.com/nightscout/AndroidAPS/files/10838097/AndroidAPS_LOG_1677485893041.log.zip)
[AndroidAPS_LOG_1677485506472.log.zip](https://github.com/nightscout/AndroidAPS/files/10838098/AndroidAPS_LOG_1677485506472.log.zip)

MilosKozak, 2023-02-27:
pls reset logging to default , do it again and give me exact time 
it's impossible to find anything in this

RalfCGM, 2023-02-27:
Ok, thats really weird. Tried again, and now it suddenly worked...

Attached the logs of the attempt that did work. Maybe theres still something pointing in a direction? I pushed the stats button 1 or 2 seconds after 19:42.

Ive compared the stats of the dana with the calculated stats of aaps, and they arent (even slightly) the same. Also attached. 
[image]
[image]

[AndroidAPS_LOG_1677523472248.log.zip](https://github.com/nightscout/AndroidAPS/files/10842982/AndroidAPS_LOG_1677523472248.log.zip)

RalfCGM, 2023-03-16:
Happened again. 
Data attached. Around 22.50 or before. 
Now tdd stats dont load anymore. Cleaning database doesnt help.

Also notice the double data in the autosens calcutions. 

Maybe related? 

[image]

[image]

[AndroidAPS_LOG_1678917065870.log.zip](https://github.com/nightscout/AndroidAPS/files/10987665/AndroidAPS_LOG_1678917065870.log.zip)

MilosKozak, 2023-03-21:
https://console.firebase.google.com/u/0/project/androidaps-c34f8/crashlytics/app/android:info.nightscout.androidaps/issues/6d7cb16597669fe6566aca6e7e224d69?time=last-seven-days&versions=3.2.0-beta1%20(1500);3.2.0-dev-i%20(1500)&types=crash,error&sessionEventKey=64170B30011E00013A9CFBE07FAE41D7_1790809612935691573

MilosKozak, 2023-03-22:
I cannot replicate this issue. and there is no error visible in logs

RalfCGM, 2023-03-24:
Thats a pity. Can i assist in some other way? Im now on dev-i. Try reporting when things happen again.

MilosKozak, 2023-08-17:
https://github.com/nightscout/AndroidAPS/issues/2397

closed 2023-08-17 by MilosKozak