nightscout/AndroidAPS #2429: The food tab in Aaps does not sync my registered foods in the Nightscout food editor

smellado, 2023-02-20:
when making change of mobile device and installing aaps from scratch, and importing the previous configuration, it seems that not all the food was synchronized from the Nightscout food editor to the food tab of Aaps.

I'm in developer mode, and I'm going with version 3.1.0.3-dev-h in Aaps and also with the development version of nightscout 15.0.0 updated last Friday, February 18.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10786444/AndroidAPS.log)

MilosKozak, 2023-02-20:
It's too vague. What does mean "not all the food was synchronized"

smellado, 2023-02-20:
Yes, all the foods that I have saved in the nightscout food base do not appear in the Aaps food tab. I have already reviewed the whole issue of synchronization with nightscout in Aaps, and everything is correct

closed 2023-02-21 by MilosKozak