nightscout/AndroidAPS #2430: Automations : Which line to delete

robertrub, 2023-02-21:
Minor bug : In automations, I long press one line in order to delete it. The delete icon arrives but no idea which line will be deleted :( As it looks, all the active automations are checked, thus those will be deleted... 

[image]

vanelsberg, 2023-02-21:
Thins this is already fixed on latest dev?
For me it shows the "to delete" checkbox on the right of the automation(s)

robertrub, 2023-02-22:
Hi Theo

I have the latest Dev (as usual ;) ). 

The delete icon is there but which line will be deleted ??? There is no information on that

robertrub, 2023-02-22:
Ok, I get it. Long press shows the del icon. Hitting the del icon shows the check boxes. Hitting again the del icon (after having checked the lines) deleted the lines...
Not very clear. I don't like to hit delete (the first tap) with items I use already checked. 
Long press on a line should make appear the del icon AND the del boxes to be checked. Then, the use would be clear.

vanelsberg, 2023-02-22:
> Long press shows the del icon

For me the bin/del icon is always showing.
To delete items:
1. I press the bin icon to show checkboxes (on the right of the screen)
2. Select one or more items to delete.
3. Press the bin icon again to confirm.

Actually this is consistent with other AAPS screens like the treatments tab?
I think the long press is a left-over from previous versions? Personally I'd like to see long-press removed as it is inconsistent and unintuitive for use?

robertrub, 2023-02-22:
There is no Del icon in the automations

[image]

There is delete item in the 3 dots menu.

I agree, long press should be inactivated and either a Del icon should be added or only the 3 dots menu / delete should stay.

vanelsberg, 2023-02-22:
Oops... for me there is....
S22 Ultra has a big screen?

[image]

robertrub, 2023-02-23:
Do you have an Automations _TAB_ or do you open it from the hamburger menu ? Our captures do not look alike, I don't see any tabs on yours

vanelsberg, 2023-02-23:
... opening from the hamburger menu.

MilosKozak, 2023-03-02:
Remove/Sort option was duplicated and not working
https://github.com/nightscout/AndroidAPS/commit/3a950bcd7f35db2d74cb17c32fbe70db0192d7a2

closed 2023-03-02 by MilosKozak