nightscout/AndroidAPS #2432: BG source NSClientV3 is missing BG values

vanelsberg, 2023-02-21:
**AndroidAPS** 3.1.0.3-dev-h
**Build**: `faee7d38a0`
**Flavor**: fullrelease / Engineering mode

**Problem description**:
Since switching to NSClientV3 on my test phone BG source NSClient started missing BG values.

**Details**
I have my dev/test phone configured with NSClient as BG source so it get's my BG data from my main AAPS phone. Together with syncing/downloading treatment from NS the test phone is "shadowing" BG and treatments from my main phone for testing purposes (no pump attached).

This setup was working perfectly fine until I upgrade Nightscout to versie 15.0.0. This version is working without problems on the main phone which uploads treatments and BG values (BYODA) to Nightscout 15.0.0 (recent dev version).

Since upgrading to NS v15.0.0 the test phone started missing BG's: soon after the phone's lockscreen activates AAPS is no longer receiving BG values. When I reopen the app it updates to the latest BG value and start updating as usual.

As this setup was working fine and as I did not change the phone's setup or settings other then updating to latest dev-h and activating NSClientv3, my conclusion would be AAPS is not properly waking on new BG values from NSClient. I am sure this has nothing to do with power savings, Google Protect etc..

**Screenprints**
_**Note that the overview is partially filled as I did a re-install of AAPS to see if this somehow solved the problem describe above**_

[image]

[image]

**Logfiles:**
[AndroidAPS_LOG_1677016311427.log.zip](https://github.com/nightscout/AndroidAPS/files/10798093/AndroidAPS_LOG_1677016311427.log.zip)

parapenT1sta, 2023-02-21:
Just one question, did you setup the NS token with "admin" role?

MilosKozak, 2023-02-25:
Did you enable websockets?
How do you upload BG data to NS?

vanelsberg, 2023-02-25:
> Did you enable websockets?

I am running my Nightscout service from Docker image *nightscout/cgm-remote-monitor:latest_dev*. So no hosting server that needs websockets to be enabled.

> How do you upload BG data to NS?

- Main phone:
is uploading treatments and BG data to NS (v15.0.0. dev) using NSClinetV3. Main phone is happy ;-)

- Test phone:
uses NSClient for BG source (download only)

MilosKozak, 2023-02-26:
BG data is available in NS?
did you try both WS ON and OFF on NSclient side?

vanelsberg, 2023-02-26:
Observations, on te test phone, NSV3 tab:
- Not seeing the regular "TICK". 
- TICKS only trailing a "UPL start/End"?

vanelsberg, 2023-02-26:
> BG data is available in NS?

Yes, there defenitely is. It is my main NS, synchonising with my daily driver AAPS

> did you try both WS ON and OFF on NSclient side?

Not sure what you mean bij "WS". WebSockets?
Where is that in AAPS settings (the client)?

vanelsberg, 2023-02-26:
Not at all convinced this is not in my NS deployment. Maybe something blocking Web Sockets between frontend and NS backend? Will investigate further and maybe go for a repro on an alternate NS deployment.

Do you know if xDrip+ (dev) is supporting NSClientV3. Could use that for a test client?

vanelsberg, 2023-02-26:
Busy, so will need some time for this so propose to park this issue. I'll keep you posted.
Feel free to close this issue for now, I'll reopen/follow-up when needed?

closed 2023-02-26 by MilosKozak

vanelsberg, 2023-02-26:
@MilosKozak 
Btw: not convinced this is a Web Socket problem as NS log output is claiming to use it...

vanelsberg, 2023-03-04:
@MilosKozak Update: Found the problem:
Switched off "connect to websockets" and it started to work (why didn't I try - you told me so :-| ):

Nightscout v15.0.0-dev (data source for BG and treatments - download only) is running NSClientV3 but somehow websockets do not work.