nightscout/AndroidAPS #2444: Re-pair needed DanaRS pump after change from 26.02.2023

Alamo04, 2023-03-03:
Before that change I only needed to pair my DanaRS once, after I get the pump over one year ago.

Now I had to re-pair my pump two times in 5 days again, last time this morning while sleeping.

Please remove this change, its annoying:
https://github.com/nightscout/AndroidAPS/commit/1fa2cb937095647843ef9febdc344b26fb6133a9

[AndroidAPS._2023-03-03_06-57-38_.1.zip](https://github.com/nightscout/AndroidAPS/files/10881583/AndroidAPS._2023-03-03_06-57-38_.1.zip)

CaptainBalou, 2023-03-03:
I can confirm that I had that too a couple of times. But not as often as on your setup. But it happened to me. During that time I had no pump replacement or so, it just happened during normal operation.

Alamo04, 2023-03-03:
@MilosKozak 
Take a  look at line 9102 in Log.
Think that was the point that the alarm and re-pair request comes up (06:56:38.479).

> 06:56:38.479 [binder:17857_5] D/PUMPBTCOMM: [BLEComm.onConnectionStateChangeSynchronized():372]: Device was disconnected SGF00019FB

Last connection to pump was a few seconds before that, in line 9073 (06:56:31.231)

> 06:56:31.231 [Thread-2629] D/PUMPCOMM: [DanaRSService.readPumpStatus():255]: Pump status loaded

MilosKozak, 2023-03-03:
i need at least 15 min of logs before and after

MilosKozak, 2023-03-03:
try dev now

MilosKozak, 2023-03-03:
https://github.com/nightscout/AndroidAPS/commit/f651f721e00b71fd2c5a1bec7f92f4a21a5a12d7

Alamo04, 2023-03-03:
Done, will report if it comes up again. 

Will test it for a week, because I had that issue twice in 5 days.

Alamo04, 2023-03-08:
@MilosKozak 
Just now happend again on workplace.

MilosKozak, 2023-03-08:
provide logs

Alamo04, 2023-03-08:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/10919170/AndroidAPS.log)
[AndroidAPS._2023-03-08_04-34-38_.1.zip](https://github.com/nightscout/AndroidAPS/files/10919171/AndroidAPS._2023-03-08_04-34-38_.1.zip)
[AndroidAPS._2023-03-08_04-34-38_.0.zip](https://github.com/nightscout/AndroidAPS/files/10919172/AndroidAPS._2023-03-08_04-34-38_.0.zip)

MilosKozak, 2023-03-08:
turning off BT adapter is not possible with Android 13 already. The code will be removed

MilosKozak, 2023-03-08:
sorry, but the problem is between phone and pump. On your place I'd try different phone first
Sadly Dana pumps don't have a way to check if encryptions keys are ok. Pump simply doesn't respond. Thus reaction of AAPS to not responding pump (after some attempts) is clearing the encryption keys

Alamo04, 2023-03-08:
The system works fine before that change, only sometimes (after 5 days today) the connection was lost and then I got an alarm, restart BT and everything was fine again. 

Can't we stay with that ? Don't need a new pairing in that case, ist's a lot faster to de-/activate BT. 

Pixel5, DanaRS, AAPS latest, Boyda

MilosKozak, 2023-03-08:
previous solution had another issues

Alamo04, 2023-03-18:
Changed to Dana-i (don't like that pump) and no re-pair comes up for the lat 7 days.
Also, if a re-pair was needed, the pairing code is mutch shorter.

So I have to use that pump now, instead of my lovely DanaRS. :(

closed 2023-03-18 by Alamo04

wootmasterslick, 2023-03-23:
Seems like I just had this problem with dana-i.
Does this screen look familiar?

[image]

Alamo04, 2023-03-25:
No, it isn't a pairing window.

Your screen shows only that your pump wasn't connected.  Perhaps empty battery?