nightscout/AndroidAPS #2447: Xdrip blood sugar

toniteh78, 2023-03-03:
Hello, with the dev-h version I have no blood glucose data from either Nightscout or Xdrip in AAPS. Can someone explain the reason?
![Uploading Screenshot_20230303_
![Uploading 
[image]
Screenshot_20230303_141630_xDrip+.jpg…]()
140646_AAPS.jpg…]()
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

toniteh78, 2023-03-03:
[image]
[image]
Screensho
[image]

parapenT1sta, 2023-03-03:
What info do you have in xDrip tab on AAPS? Was working before?

toniteh78, 2023-03-03:
> Каква информация имате в раздела xDrip на AAPS? Работеше ли преди?

On dev-e everything was working fine now it looks like this
[image]
[image]

toniteh78, 2023-03-03:
what could this mean?
[image]

toniteh78, 2023-03-03:
Is ok

closed 2023-03-03 by toniteh78