nightscout/AndroidAPS #2448: [DEV] Status Line is not showing all information

triplem, 2023-03-03:
The Statusline shows the Date of the last reservoir change as well the the age of the Battery, but not it's values. 
On the Action Screen only parts of the value are shown. In Landscape Mode this seems to be working fine.

I would prefer of the value is shown before the age, ao to be able to show the value instead of the age of both.

triplem, 2023-03-03:
The Phone is a OnePlus 8T.

MilosKozak, 2023-03-03:
screenshot?

triplem, 2023-03-04:
[image]

[image]

[image]

[image]

MilosKozak, 2023-03-04:
this doesn't have good solution. formating is done by android

vanelsberg, 2023-03-05:
Screen look good for me.
Think is is phone or phone settings specific, maybe check phones display setting like resolution en/or font size?

swissalpine, 2023-03-05:
If you select "Skin: Low resolution skin" in the settings, then the superfluous, space-eating labels like "Alter:" and "Abgegeben:" are hidden.
In the past (some years ago), these labels did not exist either. And since the information is actually self-explanatory, we could basically say goodbye to it again. Or @MilosKozak?

vanelsberg, 2023-03-05:
> If you select "Skin: Low resolution skin" in the settings, then the superfluous, space-eating labels like "Alter:" and "Abgegeben:" are hidden. In the past (some years ago), these labels did not exist either. And since the information is actually self-explanatory....

Translations may give problems, like in this case.
I think for most this is not a problem. How about just showing icons, lik eon main overview, pump/reservoir/sensor?

triplem, 2023-03-09:
Well, setting "Low Resolution" dies Work. But eg "Abgegeben" is then still shown.

#
[image]
 
Also this does not really has an effect on the overview 

[image]

Still I think that the level in the reservoir is more important then the age. Could this be swapped? Same is probably true for the battery level (well, at least partly).

Philoul, 2023-03-26:
> Still I think that the level in the reservoir is more important then the age. Could this be swapped? Same is probably true for the battery level (well, at least partly).

On low res screen, there are less information than on higher res:

With my phone for ex, I have age and level for reservoir, age and level for battery

[image]

But it's a good idea to keep levels and hide ages on low res...

aliswalker, 2023-04-21:
@Philoul without going OT as this does affect status lights on low res – Do you know why your COB doesn't display a timestamp like mine/OP? It's on a new line which for low res users means scrolling with status lights on.

[image]

Yours both also show TBR % vs U/h which was proposed as a setting under #2200

Philoul, 2023-04-21:
1) The time below COB value is only visible in dev branchs, not in master (I just hide it on my own version).

2) The format of TBR is directly linked to the pump driver and how the pump manage this information:
- for ex with Insight pump, the driver send a % to the pump for standard tbr, and managed the extended tbr by a square bolus. That's why I see in Overview a % when it's a standard tbr is running, and an absolute rate value when an extended tbr is running.

So if your pump driver send to your pump an absolute rate when a tbr is set, then you will see the absolute rate in overview... (I think it's to be able to check directly with your pump UI if both informations are consistant)

closed 2023-05-15 by MilosKozak