nightscout/AndroidAPS #2449: AAPSClient missing entries from AAPS

bchris21, 2023-03-05:
Both on Android 12
NS Dev15
AAPS DEV-h 
Logs attached.
Websockets ON on both.

Treatments entered by AAPS on 10:23 (25gr carbs), bolus 0.7 at 10:54 and bolus 1.1 at 12:19 not reflected on AAPSClient.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

streethawk22, 2023-03-06:
I have the same experience. Not all comes to the AAPSclient. The comunication problem is also by the switching profile on AAPSclient App, not always get the command trough the NS to the Master App.

MilosKozak, 2023-03-06:
could you check if there is WS reconnection around missing period?

bchris21, 2023-03-06:
Manual NSClient restart and AAPS yes; while troubleshooting in order to see if a restart will finally push the missing treatments.
Now if there was an instant network disconnection I cannot confirm.

Let me know if I can assist troubleshooting in any way.

Without a fully working AAPSClient and NS being able to set temp targets, enter carbs etc it's very challenging to remotely control our kid while being at school. As a workaround TeamViewer/Airdroid is used to remotely control (also SMS Commands).

gpgerrard, 2023-03-08:
Having the exact same problem with AAPSClient 3.2.0-beta1 and NS15 using v3 API.  Websockets on. Treatments appearing OK on one AAPSClient instance but not on another. Will keep an eye out for WS reconnections.

CaroGo, 2023-03-12:
Same observation...entering carbs or tt in nightscout manually or via ifttt is not displayed and taken into account in AAPS. (Infos arrive in xdrip and nightscout).

jumalakarhu, 2023-03-29:
I have a similar experience, but the opposite way. AAPS client is set to _not_ upload treatments and profile but to download treatments and profiles. Still when I enter 1 g carbs on AAPS client this is sent to Nightstcout side and further to AAPS. 
Also - Profiles from AAPS are not uploaded to NS and not sent to AAPS client. Tested on new install of AAPS client and another with total wipe (force stop, clear cache and storage and folders) before reinstall. If I set 
All boxes in AAPS nsv3 sync ticket except receive/backfill cgm data. Have done full sync on both AAPS an client to no avail. As it is now AAPS client does not have valid profile (
AAPS Android 12, and client android 11 dev-i, NS dev15 websockets ticked.

MilosKozak, 2023-03-31:
> Same observation...entering carbs or tt in nightscout manually or via ifttt is not displayed and taken into account in AAPS. (Infos arrive in xdrip and nightscout).

you didn't read limitations in release notes

MilosKozak, 2023-03-31:
> Having the exact same problem with AAPSClient 3.2.0-beta1 and NS15 using v3 API. Websockets on. Treatments appearing OK on one AAPSClient instance but not on another. Will keep an eye out for WS reconnections.

it has been improved in dev-i

jumalakarhu, 2023-03-31:
> 

Update: Seems like a restart of dynos on Heroku and then a full sync from AAPS have fixed issues with profile sync. Still get carb entries entered in AAPSClient uploaded to NS and AAPS even though sync from AAPSClient is not enabled.

MilosKozak, 2023-05-14:
this has been improved. open new issue if this re-appear

closed 2023-05-14 by MilosKozak