nightscout/AndroidAPS #2451: Unable to set english as language [3.2.0-beta1]

yhonatanb20, 2023-03-07:
The beta has again the problem that appeared on earlier versions (3.0.0 and earlier) and was fixed with 3.1.0, which makes AAPS fixed on system language (setting english as system language makes it some wierd half-half hybrid). Please fix that, it was wonderful to be able to set english in 3.1.0.3, would greatly miss this feature...

MilosKozak, 2023-03-08:
sorry I don't know how to improve this. Basically setting different language then Android is cheating and may not work in all situations. Paste here screenshot so I can at least test

szantos, 2023-03-15:
Apart from the system language, using AAPS in English has the tremendous benefit, that you can easily find corresponding parts of the documentation, error messages, etc.

Currently the language chosen in AAPS (English) and system language (other) cause a mess of both languages. Units and 12/24h time also affected.

IMHO:
units should = language chosen in AAPS
time format should = system language

MilosKozak, 2023-03-21:
fixed by https://github.com/nightscout/AndroidAPS/commit/75d34955b7aba0782014631cd8656b7a67153186

closed 2023-03-21 by MilosKozak

yhonatanb20, 2023-04-30:
Still experiencing this issue.
3.2.0 Dev-i 27.4.23: the issue has been fixed mostly, but on some particular screens (I found it on treatmets and on DASH screens) system language (here hebrew) is still used albeit english set in prefrences:
[image]
[image]
[image]

yhonatanb20, 2023-05-05:
@MilosKozak could you fix that?

MilosKozak, 2023-05-17:
fixed by https://github.com/nightscout/AndroidAPS/commit/7faa34aa50d7deca20b344e23f6a136b30586501