nightscout/AndroidAPS #2459: [3.2.0-beta1] Random crashes

mxsrm, 2023-03-13:
I encountered random multiple random crashes (frequences like once or twice a day) with no evident reason in the [logfiles](https://pastebin.com/raw/zZnvsSuT). After 07:45:45.106 the app was just suddenly dead and only started working again when I opened it manually at 12:28:30.

Using Dexcom G7 with patched app as BG source, Nightscout v3 as client to upload values and a DanaRS as pump.

MilosKozak, 2023-03-13:
unresolvable. but because there is no other reposts like this i'd guess it's phone . without anything suspicious in logs it seems android is killing app

closed 2023-03-13 by MilosKozak

sdolenc, 2023-06-20:
I'm getting random "crashes" in the latest dev branch as well, but there's no indication of any runtime issue. I think you're right that the phone is killing the app. I don't think the pixel 4a killed the official release as much. (or maybe it did and the app re-launched somehow)

I'll play with phone settings to see if I can keep it running longer. Does anybody have any tips?

I may also use an [automate](https://play.google.com/store/apps/details?id=com.llamalab.automate) script to relaunch AAPS if needed

Thanks for everything Milo! I can't express how much you've helped our family! We just found out about the donation [page](https://ko-fi.com/diabetesone) and will be making annual contributions to help support the project