nightscout/AndroidAPS #2460: Androidaps skips over latest xdrip measurement

gitthiy, 2023-03-14:
Reporting bugs
--------------

AndroidAps is getting bg values from xdrip broadcast 
Xdrip provides value each 5 minutes 

On AndroidAps you can see latest value when xdrip gets latest value, this latest value is displayed for a short moment but will be ignored with previous value

Attached version Infos of androidaps / xdrip
Displayed bg and received xdrip values 

[image]

[image]

[image]

[image]

MilosKozak, 2023-03-21:
most of the code is changed for 3.2
please re-test with 3.2

MilosKozak, 2023-05-14:
closed for inactivity

closed 2023-05-14 by MilosKozak