nightscout/AndroidAPS #2463: 3.2.0 dev i Combo Pairing

Duester64, 2023-03-19:
Native Combo driver. Can not put in the 10 digit pairing code after APS found the bluetooth. After input the code there is no enter or send button so finally the pump closes the connection on timeout. Tried it many times, always the same behavior. Pleas add the necessary button. Otherwiese i have to continue with ruffy. Thanks in advance.

andyrozman, 2023-03-19:
That is weird... It works for everybody else... including me... Are you using very small phone (which might hide the button)?

Duester64, 2023-03-19:
IUsing  S21. Perhaps you can show a screenshot where the button is shown. I expected the button to show up after entering the last of the 10 digits but nothing.  Any other suggestions?

andyrozman, 2023-03-19:
I think guys on discord (in combo-v2 group) and developer are already looking into it.

Der-Schubi, 2023-03-22:
This will be fixed with https://github.com/nightscout/AndroidAPS/pull/2464
The input field has a fixed with in the moment, just put your device in landscape mode to press the ok button, after entering the pairing code.

closed 2023-03-25 by MilosKozak