nightscout/AndroidAPS #2467: Problems pairing the Glucomen day pump AAPS-dev

mtanjagithub, 2023-03-22:
Reporting bugs

at 10 am I tried to start a new pod. At step two, AAPS remains in screen 2/6.
Build versoin: 294f3cb789-2023.03.15
[AndroidAPS_LOG_1679479924763.log.zip](https://github.com/nightscout/AndroidAPS/files/11038879/AndroidAPS_LOG_1679479924763.log.zip)

mtanjagithub, 2023-03-22:
the same problem came up again.

at 18:34:48.379  I tried to start a new pod. At step two, AAPS remains in screen 2/6.
Build versoin: https://github.com/nightscout/AndroidAPS/commit/294f3cb789193b971da2dfd5ca629bda999a497b-2023.03.15

[AndroidAPS_LOG_1679506713006.log.zip](https://github.com/nightscout/AndroidAPS/files/11043618/AndroidAPS_LOG_1679506713006.log.zip)

MilosKozak, 2023-03-23:
This error occur when pairing was not successful ie Android failed to pair device or there was no user response on pairing attempt

closed 2023-03-23 by MilosKozak