nightscout/AndroidAPS #2473: 3.2.0-dev-i TDD screen shows AndroidAPS logo in the background, making it difficult to read TDD stats

dv1, 2023-03-23:
On a test KKM phone, I saw that the TDD screen shows the AndroidAPS logo in the background, which isn't good, because the white portions of the logo make the white TDD stats text hard to read.

Here is an example. [image]

Previous AndroidAPS versions kept the background fully black in this screen, and it would be good to do the same in this version.

vanelsberg, 2023-03-23:
Not seeing this on latest dev builds (3.2.0-dev-i, 1920bc7ce5)
Moreover, I do not recognize the layout showed in the screenshots above so I think your are building from some alternate AAPS fork?

dv1, 2023-03-23:
@vanelsberg  What does the screen look like for you? Post a screenshot please.

vanelsberg, 2023-03-24:
[image]

parapenT1sta, 2023-03-24:
I don't have the same issue but was one like this one for EROS but already fixed.

https://github.com/nightscout/AndroidAPS/issues/2390

dv1, 2023-03-24:
@vanelsberg That is a different screen. What you show is the Statistics screen. I am talking about the TDD screen available through the Actions tab.

parapenT1sta, 2023-03-24:
@dv1 I can confirm! I also have the same issue going through Actions - TDD.

Maybe the fix is the same as the one used for EROS.

vanelsberg, 2023-03-24:
> I am talking about the TDD screen available through the Actions tab.
For there is no such screen fro me. Are you building 3.2.0-dev-i from https://github.com/nightscout/AndroidAPS/tree/dev?

parapenT1sta, 2023-03-24:
> > I am talking about the TDD screen available through the Actions tab.
> 
> Nope. For me there is no such screen fro me. Are you building 3.2.0-dev-i from https://github.com/nightscout/AndroidAPS/tree/dev?

[image]

parapenT1sta, 2023-03-24:
Using ComboV2 driver

closed 2023-03-25 by MilosKozak