nightscout/AndroidAPS #2475: AAPS 3.2.0 dev-i is not uploading to NS smoothed values from G7 patched app.

emmatovar27, 2023-03-26:
Reporting bug

I changed today from BYODA G6 to the patch app of G7. 

It looks like AAPS needs to upload the smoothed values. Instead is uploading the raw values to Nightscout. 

[image]
[image]
[image]

I'm using version AAPS 3.2.0 - dev-i Build:003f323d4c-2023.03.25
I'm using the plugin that uses the API v3 and NS version 15.0.0

Here are the logs too. 
[AndroidAPS_LOG_1679817337271.log.zip](https://github.com/nightscout/AndroidAPS/files/11071036/AndroidAPS_LOG_1679817337271.log.zip)

blaqone, 2023-03-26:
Maybe add a switch to allow uploading smoothed OR raw data to NS.

BUT then again it is the smoothed data you rely on if you enable that smoothing plugin - so I think the smoothed data must be written to NS.

Pamalosebi, 2023-03-26:
yes, please. I guess it would also render services like [https://autotuneweb.azurewebsites.net](https://autotuneweb.azurewebsites.net) useless.

blaqone, 2023-03-26:
> yes, please. I guess it would also render services like https://autotuneweb.azurewebsites.net useless.

Autotune has nothing to do with smoothing values.

MilosKozak, 2023-03-26:
this is by design. AAPS doesn't modify BG values. But it can does smoothing for internal use and rendering

closed 2023-03-26 by MilosKozak

blaqone, 2023-03-26:
This makes no sense - the smoothed values is what you "work" with and why should the NS values not represent what you daily calclulations are based on ?

Pamalosebi, 2023-03-26:
> > yes, please. I guess it would also render services like https://autotuneweb.azurewebsites.net useless.
> 
> Autotune has nothing to do with smoothing values.

Are you sure about this?

As far as I know it uses CGM data from NS. How else would it find possible UAMs?

It may be that "Autotune has nothing to do with smoothing values", but I am talking about the calculations that are done with the raw data available in NS. And if these are not smoothed (as it should be) then I don't see how these services are supposed to work with the data. Also when generating reports and so on...

blaqone, 2023-03-26:
> Are you sure about this?
>
> As far as I know it uses CGM data from NS. How else would it find possible UAMs?
>
> It may be that "Autotune has nothing to do with smoothing values", but I am talking about [...]

Yes this is indeed correct!

MilosKozak, 2023-03-26:
AAPS even keeps original values in DB. data is smoothed just before use and only in memory