nightscout/AndroidAPS #2483: [3.2.0-beta1] Profile helper 'Clone' reports Error in IC Values

Vigilo-Rob, 2023-03-30:
When an existing working local profile is cloned, it generates a DefaultProfile <datetime> local profile.

When viewed in the LP tab - the profile is highlighted in Red, and has invalid entries (different to the original profile) in the Basal rate fields etc

Changing from a working profile to the newly cloned DefaultProfile shows an error at the bottom of the screen saying 'Error in IC Values'

build: b4e5b01ea3-23.03.03

[image]

[AndroidAPS_LOG_1680169503885.log.zip](https://github.com/nightscout/AndroidAPS/files/11110157/AndroidAPS_LOG_1680169503885.log.zip)

MilosKozak, 2023-03-31:
sorry explanation is not enough to be able replicate the issue.
put here step by step screenshots what you did and where are the differences

Vigilo-Rob, 2023-03-31:
No problems - see attached Zip for screenshots.

Those entitled 'Working DAI' etc are the profile that is cloned and shows my current profile settings.

Those entitled 'Error DAI' etc are the cloned profile once its created.

The 'Profile Helper' screenshot is before I press the 'Clone' button.

[Profile Helper Error Details.zip](https://github.com/nightscout/AndroidAPS/files/11119914/Profile.Helper.Error.Details.zip)

MilosKozak, 2023-04-04:
working profile has a name 40pct fiasp.
Failing profile is DefaultProfile
thus it's not a clone of "40pct fiasp"

Vigilo-Rob, 2023-04-04:
DefaultProfile is the name it allocates to the profile when you press the 'Clone' button - not sure what you expect to happen? My guess is it can't clone a profile with the same name else there would be a duplicate profile hence uses DefaultProfile and Datetime to identify the new profile

The problem appears to be the values created in the profile itself - look at the IC ratio on the cloned profile - its shows an IC ratio of 23370203 with 7 different ICs

The IC ratio from the source cloned profile is 7.3 and a single time entry.

MilosKozak, 2023-04-04:
but this is not your current profile. it kind of recommendation where to start if you have no profile
what values did you put in?

Vigilo-Rob, 2023-04-04:
Current profile I used to clone is 40pct fiasp (thats what I'm actually using for the loop)

[image]

Vigilo-Rob, 2023-04-04:
Right - light finally dawns - (thank you)

So it's not possible to clone the active profile, and the default profile has out-of-range entries to force the user to enter their own profile rather than accepting a default value.

Sort of makes sense - apologies for asking so many questions.

closed 2023-04-04 by Vigilo-Rob