nightscout/AndroidAPS #2484: [AAPS 3.2.0-beta1] Carbs get duplicated, even three times the entered amount

ledrexter, 2023-03-30:
Hi,

I'm using AAPS 3.2.0-beta1
build: 616b4650ba-2023.03.07
phone : Xiaomi 11T Pro
Android version: 13 TKQ1.220829.002
MIUI version: 14.0.1.0(TKDMIXM)
Nightscout:  15.0.0

Two consecutive day I have COB three/two times the amount I entered.

29.03.2023 at 12:53 I enter **46 grams** of carbs but then I ended up with  **3 * 46g = 138g COB**
30.03.2023 at 13:02 I enter **50 grams** of carbs but then I end up with **100g COB**

I thing it could be related to slow internet connection and some how Nightscout, but I'm not sure. I do have launch at the same place everyday and yesterday and today are the first sim I observe this kind of issue.
 

I am attaching log files as well as screenshots:
AndroidAPS_LOG_1680084846821.log.zip is for the 29.03.2023 at 12:53
AndroidAPS_LOG_1680171136301.log.zip is for the 30.03.2023 at 13:02

Attachments:
https://drive.google.com/drive/folders/1norDDVYO3Aj92GCxJceUlzSbxRXxnO90?usp=sharing

Kind regards,
Petar Harizanov

MilosKozak, 2023-03-31:
the crucial parts in logs around time of occurence are missing for both cases
generealy duplication is usually caused by xdrip synchronization, turn if off and test again

ledrexter, 2023-04-04:
I will need to check if there is a trace in the other part of the log and I will let you know.
By the way it has not happened again since then, although I have not touched the aaps/xdrip settings.

MilosKozak, 2023-04-10:
> I see the same problem intermittenly. Sometimes double, more seldom tripple, but it has happened. I dont use xdrip. aaps dev-h NS 15.0 built on ac1a7d1bd880426b96a6acb3d594d26f1a76a730
> 
> https://drive.google.com/file/d/1lfBBmNfUkQ7lr4y__pp7E9SLVkT1w9ej/view?usp=sharing

time? screenshots?

MilosKozak, 2023-05-14:
logs don't have 10:33

kristinsmotz, 2023-05-16:
I have seen duplicate carb entries as well, for us they happen after a bolus is cancelled partway through. Not every single time, but often enough not to be a fluke, so when I cancel a running bolus I make sure to watch for duplicate COB the next few minutes. Typically when we cancel a bolus it's because of remembering (or kid informing of) planned activity right after meal time, which tend to cause lows if we have too much IOB. 

My understanding is that this is what happens: (Bolus at 17.33 in attached log, I've tried to interpret what I found in the log that seemed relevant, with my comments added).
17.33.14: Bolus calculator used for 50 carbs, suggesting a bolus of 3.4U, posting insulin amount to treatment (including it in IOB). 
17.33.30: The bolus is cancelled. 
17:33:31.172: Original 50 carbs (?) posted to treatments/COB. 
17:33:31.181: Bolus amount updated with correct delivered amount, (original insulin treatment deleted? and new insulin treatment with correct amount posted)
17:33.31.188: Extra 50 carbs posted to treatments/COB, so now COB=100g
At that point I exported logs then went back to delete one 50Carb entry so SMB wouldn't keep using 100COB for calculations, as kid was running out to play... 
We are using AAPS and AAPSClient, 3.2.0-dev-i.

[AndroidAPS_LOG_1683819235198.log.zip](https://github.com/nightscout/AndroidAPS/files/11485735/AndroidAPS_LOG_1683819235198.log.zip)

MilosKozak, 2023-08-16:
resolved by https://github.com/nightscout/AndroidAPS/commit/fb94393c1c53ea72ff3c8286bcefebd1d16bd637

closed 2023-08-16 by MilosKozak