nightscout/AndroidAPS #2488: [question] Releases to f-droid?

freed00m, 2023-04-02:
Hi,

Just wanted to ask why no releases on f-droid? Or is it in consideration?
The project could gain even more popularity.

Or host a custom repository. 

It's probably still doesn't have multiple release channels for rc and alpha.

But for many people it would be helpful to get updates via notification->click install.

On non rooted phone, the updates can't be set to install automatically so I don't think there is the danger of having broken app unnoticed.

ArthurusDent, 2023-04-02:
> The app is not provided in Google Play - you have to build it from source code by yourself for legal reasons.

Source: https://androidaps.readthedocs.io/en/latest/

The same probably applies to F-Droid.

freed00m, 2023-04-02:
I know how to get it / build it.

And I know why it's not on google play.

Fdroid is a a great way to diatribute foss and open source software on Android.

I wanted to start discussion about it.

The fdroid would provide the apk if the maintainers would either be manually releasing the bundle or github action script taking care of signning it and pushing it fdroid.

MilosKozak, 2023-04-03:
like said above. it's legal reason

closed 2023-04-03 by MilosKozak

freed00m, 2023-04-03:
@MilosKozak the doc says "The app is not provided in Google Play - you have to build it from source code by yourself for legal reasons."

I am not concerned about Google play, I am talking about F-Droid which has none such issues because it is meant for open source software.

ArthurusDent, 2023-04-03:
There are national laws for medical apps or medical devices. It is very likely that AndroidAPS would fall under such laws. Whether one would be able to get out of liability for damages caused by AndroidAPS simply by referring to the GPL is highly questionable. The developers understandably have little interest in having this question settled in court, and have therefore chosen a form of distribution that leaves little doubt that the patient himself is responsible for any damages resulting from the use of AndroidAPS.

If you are still not convinced, please contact a real lawyer (not Google).