nightscout/AndroidAPS #2490: AndroidAPS doesn't handle backdated profile switches correctly

Kenoubi, 2023-04-04:
AAPS 3.1.0.3.  The Profile Switch dialog has an Event Time field, and this is uploaded to Nightscout, but it appears to be totally ignored by AAPS itself; the profile switch event always shows up on the main graph at the time I entered it, and the different values in the profile aren't used when (re)calculating IOB, COB, etc.

I need this because I need to use different sensitivities when I'm sitting quietly in a chair (or sleeping) vs on my feet.  I've also recently been ill and needed to take almost twice as much insulin, which I used a % profile switch for.  In cases like this, I may forget or be unable to enter the profile switch at the correct time.  Obviously it's impossible for AAPS to go back in time and change the decisions it already made, but I need it to re-do the calculations using the correct profile because otherwise my IOB and COB will continue to be wrong after I notice the wrong profile was used and fix it.

This behavior is surprising to me, because other, non-profile settings like min_5m_carbimpact cause a full recalculation, so AAPS obviously already has the logic to do this; it's just not using it for profile switches.  I assume it's clear enough what I'm saying the current behavior is and how I'd like it to change, and why, but I'd be happy to answer any questions.

MilosKozak, 2023-04-05:
here is the answer
https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html#version-3-0-0

Kenoubi, 2023-04-05:
I think you are talking about this?

> Profile switch behavior change. Now is distinguished between Profile Switch (something that user wants) and Profile change (when change is executed by pump) @MilosKozak @Tebbe

Distinguishing these events is necessary and desirable (to correctly determine what insulin was delivered during periods of no connectivity to the pump, for example), but the current behavior results in AAPS making dosing decisions that are unexpected and lead to worse BG control.  Again, I don't expect it to change history of what insulin was delivered, but my IOB and COB are a less accurate reflection of what is happening in my body when they are calculated with IC, ISF and "normal" basal (what basal would have been if AAPS hadn't been changing it) taken from the profile that was in effect on the pump, rather than the profile that I've told AAPS should have been used for that time period.

True, if the pump didn't enact the profile change at the time of the profile switch, then any on-pump calculations of IOB and COB will unavoidably be wrong in that same way, but it seems to me anyone using AAPS can't be relying much on such on-pump calculations anyway.  (In practice I'm using Omnipod Dash pods, so the pump has no UI, so I can't even see any of this, and I don't think it has on-pump IOB / COB calculations anyway - the pod is basically just an insulin delivery device and the on-pod smarts are just so it does something closer to what I wanted during any periods it can't talk to the device controlling it.  I doubt the pod even stores IC or ISF values (why would it?), though it does store basal schedules.)

Philoul, 2023-04-05:
In database you only have TBR records, so if you want to change the profile in the past, you will also have to re write all the history of tbr (create fake tbr for periods without tbr running if profile has changed, all percentage should be updated...).
It's I think too complicated, risks are too import, and it's not the history of your loop (decision made by the loop was done with real active profile and not a new expected profile ipdated later).

When you update your profile. it should be little steps by little step, starting with a secure profile (probably very few hypo. but probably too many hyper at the beginning), and slowly increase aggressivity to increase TIR...

Kenoubi, 2023-04-05:
I mean, it's already setting a TBR on the pump even when the value is unchanged (`skip_neutral_temps = false`).  (It doesn't do that when it can't talk to the pump, of course.)  I don't think it would be much of a stretch to just have a TBR record in the database for every time interval, indicating how much insulin was actually delivered.

The *outputs* of the algorithm are, of course, based on the current profile (along with other inputs), as you said.  However, some of the other *inputs* to the algorithm are precomputed based on the profiles that were in effect in the recent past, notably including COB (because this is decreased over time based on the deviations in BG between what the model expects under the assumption that carbs don't appear, and the actual observed BG) and IOB (because this is the net insulin delivered (discounted for how long ago it was delivered) *relative to normal basal for that period*).  It's these *inputs* being off that cause the algorithm to make decisions one would rather it didn't.  For example, if it thinks I have a ton of IOB, it may not deliver more insulin even if BG is high and increasing, because it expects BG to start dropping a lot very soon (even though it keeps being "surprised" when that doesn't actually happen).

My insulin sensitivity varies *extremely dramatically* by what activity I'm doing.  It's at least double just to go from sitting quietly in a chair to walking around, and running / carrying heavy things / etc. can easily make it 3-4x.  Temp targets don't work, because even what would normally be a pretty small amount of insulin, starting from a pretty high BG, can make me hypo; zero temping is too slow to fix it at that point.  But if I just tripled all my sensitivities, my BG would be super high most of the time because any time it went up, AAPS would have an extremely exaggerated opinion of how much the amounts of insulin it delivered would lower my BG.

Profile switches seem like the tool for this, and I'm using them.  And they actually work!  But if I forget to do one at the right time, now I have two problems - whatever questionable decisions were made in the past based on the wrong profile (which are too late to fix) and whatever questionable decisions will be made now and in the future based on *still* assuming the wrong profile for the past (which are avoidable, and what this bug is about).  It's bad enough I've actually created fake "profiles" called "Burn IOB" and "Burn COB" with absurdly high basals and absurdly high IC ratios respectively, so I can pause the loop, set a manual temp basal, and use a timed profile switch to force AAPS to burn off the extra IOB / COB.  (Having too *little* IOB / COB is not as much of a problem since I can always just add backdated insulin (using the "Do not bolus, record only" checkbox) or carbs, though it can be pretty hard to get the timing right.)  Obviously this is extremely difficult for me to do, compared to AAPS just using the profile I want to recalculate the history.

Kenoubi, 2023-04-05:
By the way, @MilosKozak, if this behavior is deliberate (I had assumed it wasn't, for the reasons I've already stated), is there a single or small number of places in the code where the decision of when to start applying the profile switch to historical calculations is made?  I'm perfectly comfortable maintaining a small patch against my own copy of AAPS.  I'm looking into this myself, but maybe it would be a lot easier for someone more familiar with the code to tell me where it would be controlled.

Thanks!  And thanks for your work on AAPS, it's still a massive accomplishment whether or not I like every detail of how it works.

MilosKozak, 2023-05-14:
doing such things would make the code very complicated, behavior would not be easy to understand by common people and would add more possible points of wrong use of AAPS. if you need it add it to your private version. 
do not forget to change version to something like 3.2.0-dev-custom

closed 2023-05-14 by MilosKozak