nightscout/AndroidAPS #2494: [dev-i] No backfilling of treatment data to Nightscout if server isn't available

swissalpine, 2023-04-08:
I back up my FC and AAPS data to a Nightscout installation on my NAS. At night I turn it off to save power.
At 6 o'clock xDrip+ syncs all blood glucose data, but AAPS not my treatment data.

[image]

NSClientV3: websockets deactivated
Nightscout: 15.0.0

MilosKozak, 2023-04-09:
full sync helps?

swissalpine, 2023-04-09:
Thanks for coming back.
Yes. After a full sync the missing data (tbr changes) are there ...

MilosKozak, 2023-04-17:
i'd need to know error received when server is off. could you provide logs?

MilosKozak, 2023-04-19:
that's correct.
I'd need log from time when you turned on your NS but data was not uploaded

swissalpine, 2023-04-19:
WLAN on at 06:01:45.377, NAS starts at 6:00 ...

06:05:17.011 [RxCachedThreadScheduler-42] D/NSCLIENT: [NSClientV3Plugin$onStart$7.accept():223]: ● RUN Starting next round MAIN_LOOP

[AndroidAPS._2023-04-18_02-00-13_.0.zip](https://github.com/nightscout/AndroidAPS/files/11277157/AndroidAPS._2023-04-18_02-00-13_.0.zip)

BTW After seeing this logs I disabled open human uploader because it seems to fail every time ... another issue? Login is correct ...

MilosKozak, 2023-04-25:
https://github.com/nightscout/AndroidAPS/commit/4c28de0b64fdf588702bd2636d7f4b4c123a872b

closed 2023-04-25 by MilosKozak