nightscout/AndroidAPS #2495: Feature request: Additional/optional data entry e.g. heart rate, temperature and meal descriptions

Snowworm3000, 2023-04-09:
I think it would be useful to have some more optional data entry possibilities. Having more data points to review could make future treatment decisions safer. This isn't a proposition for changes to be made to the current algorithms, rather a way to potentially form the foundation of newer ones.

For example, a temperature sensor could allow the loop to react to sudden changes e.g. taking a hot shower which shortly increases insulin sensitivity for me. 

Heart rate could be taken from Google Fit or another fitness app to be used when predicting changes in ISF. 

I have some more details on carb entry which I may describe in a separate issue.

There could also be the concern that any one of these sensors could be inaccurate, so there could be a config menu to select which sensors/data entries should be used in open or closed loop mode individually.

I'm looking to contribute myself, but I just wanted to ask if any of this seemed feasible without major changes to the database/nightscout integration. Does it matter if there are inconsistencies with Nightscout?

Snowworm3000, 2023-04-10:
I might as well point out this comment https://github.com/nightscout/AndroidAPS/issues/269#issuecomment-768206150
I don't know if the migration to Room has taken place yet. Would this still be a concern?

buessow, 2023-04-15:
I'm planning to add support for storing your heart rate and using that for automation tasks. That will allow you to automatically do a TT etc when heart rate is elevated during sport.

buessow, 2023-04-17:
I've written down my plan. Comments welcome: https://docs.google.com/document/d/1RwsHYN0xWBZ6WfaFFSqMxqc4Jl4gLZJOipbpBVzyiQk/edit?usp=sharing

mtanjagithub, 2023-04-17:
I really like this idea. i am an athlete. train 10 to 20 hours a week with different workouts from endurance training (4 to 6 hours) to heavy interval training HIIT (2 hours). I notice the change in insulin sensitivity enormously, especially at night after a workout. the difference with a rest day can then very quickly add up to 25% in my AAPS profile BAS and ISF. I love to help if you need data (trainingsdata, Wattage, duration, BPM) for this

buessow, 2023-04-17:
Interesting, for me the loop algo picks up the change in insulin sensitivity after sports pretty good automatically but it totally overreacts during sports. Therefore, I switch the profile and that works quite well based on heart rate. What device are you using to measure heart rate?

Snowworm3000, 2023-04-17:
Which APS algorithm are you using? 

I often have to really lower my bolus doses to manage post exercise meals. It's probably just different metabolisms, I'm slightly underweight which could affect things like this.

buessow, 2023-04-17:
I'm using _Sensitivity Oref1_ and _OpenAPS SMB_. I guess metabolism is different. Right after sport, I sometimes skip bolus or reduce manually.

mtanjagithub, 2023-04-18:
I also use Oref1 and OpenAPS SMB. During exercise I use a temporary goal of 5.8 instead of 5.5. As a result, I turn off SMB during my workout and only get temp basals. During my workouts I increase my profile from 100% to 140 to 150% and I eat 70 to 90 grams of KH/hour. After training I reset my profile to 100%

During exercise, your body burns KH depending on the intensity. When you exercise at 75% of your threshold, your body globally uses 25% carbs and 75 body fat as fuel. If your intensity increases (you start running or cycling faster, for example), your body will use more carb as fuel. When you go all out your body will use 100KH as fuel and 0% fat.

So, it makes sense if you keep eating the same, or if you don't eat at all, you'll need less insulin because your body uses the carbs for fuel. As a result, the body will use its stores of KH. When you exercise for a long time and intensively, such as a competition, your body can sustain this for a maximum of 1 to 1.5 hours and your supplies will be exhausted. This is why I eat so many carbs during my competitions and training.

ISF works reasonably well for me when I am more sensitive to insulin after a workout. ISF adjusts my profile. And that's fine. Except, if I go to sleep 2 hours after my dinner and I eat protein before I go to sleep, ISF gives me a correct of 115 to 125% resulting in a mid-night hypo. This is because the protein intake is slow and ISF thinks I am more resistant then.