nightscout/AndroidAPS #2504: DANA i - wrong IOB after Bolus delivery (could lead to overdosing)

Meinosaurus, 2023-04-15:
AAPS 3.1.0.3 - Dana i - Motorola g9 Play Android 11

After i initiated a bolus of 5 iE the delivery went through. But afterwards the IOB in the frontpage of AAPS was still at the same level as before the bolus. As if they have not been delivered. And it stays there until you deliver another bolus.
After i started the delivery i turned the screen off. (maybe that plays a role?)
It happens quite often and as i have heard am i not the only one having this possibly dangerous issue.

I will attach the log files and en extra Txt file where i copied out the relevant lines from the log file so you dont have to search so much and you see the time stamp right away.

[AndroidAPS_LOG_1681584644647_log.zip](https://github.com/nightscout/AndroidAPS/files/11240497/AndroidAPS_LOG_1681584644647_log.zip)

Thx for your awesome work!!! :)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2023-04-16:
well, did you read the message on screen? you must MANUALLY check real delivered amount because if the phone lost the connection to the pump there is no way to get the information about issued bolus. Ususally it's enough to force new connection to the pump and the history is read

closed 2023-04-16 by MilosKozak

Meinosaurus, 2023-04-16:
Yea of course. But I was not away from the Phone. This message pops up a lot of times even when I am next to the phone. And reading the history did not work. It only worked when I initiated another bolus. Restating the phone, and turn bluetooth on and of had no effect…
Like I said, it’s weird and it appears to be a serious issue which i think can lead to dangerous situations. It did. once for me, when I double Bolused my breakfast, because I trusted the IOB…which never was a problem with my Combo, but there seems to be a problem with the Dana.

MilosKozak, 2023-04-17:
the history loadable from dana tab is not used for bolus data. 
you must press BT icon to force new connection to dana pump

Meinosaurus, 2023-04-18:
OK, will try that the next time this happens Thx. Though i think i have tried this before without any success…Anyway, i think there seems to be some kind of problem with the Dana Bluetooth Integration and it would be cool if that is taken seriously. Like i said, i am not the only person having these kind of issues. Dont you think so? Or do you think its only happening to me?
I know you are all doing this awesome work in your free time and i really appreciate that a lot! I just wanted to help and point out that there seems to be this problem that some people have, which might lead to dangerous situations…

MilosKozak, 2023-04-19:
BLE Dana is most stable pump from connectivity point view. Issues are always caused by phone

Meinosaurus, 2023-04-19:
Ok THx, then it must be Problem of the moto g9 play i guess.