nightscout/AndroidAPS #2509: New Feature

bobhome007, 2023-04-20:
Create a new event type in automations so that a trigger can be set on accelerating or decelerating BG.

Such that AAPS will store a new metric called BG Acceleration that allows users to identify the absolute rate of acceleration of a BG rise or deceleration of a BG Fall.

Where BG acceleration/deceleration is the rate of change of BG velocity (speed) of rising or falling (second derivative)

Rationale/Reasoning:

Acceleration/deceleration of BG can be calculated from the deltas but it is not possible to do such a calculation in automations.

Therefore it is not easily possible to get an absolute acceleration/deceleration magnitude of BG.

This is vital if profile % increases are being set in automations in AAPS.

ga-zelle, 2023-05-12:
In the context of autoISF I developed the first extensions to automation. So far 3 new actions were created but next is a trigger for acceleration just like your request. However, the significant difference is that I will use the smoothed acceleration as calculated in autoISF already. Deriving acceleration from the difference of deltas is far too noisy and therefore unreliable to derive any action based on it.

ga-zelle, 2023-05-17:
When I started th ework I had second thaughts. I'll develop an automation based in certain values of a weighting factor for the acceleration. That's a bit different from what I said before. Sorry I mixed it up in my first note. AutoISF modulates ISF based on the acceleration or deceleration. So you can use that already now by switching from regular AAPS to the version with autoISF included. For details check my Github repo in ga-zelle/autoISF.

bobhome007, 2023-06-16:
> When I started the work I had second thaughts. I'll develop an automation based in certain values of a weighting factor for the acceleration. That's a bit different from what I said before. Sorry I [...]

Will this "automation event/trigger" make it into live AAPS and is it possible to use it without having the AutoISF algo set in preferences?  

The enhancement requested is to make the acceleration/deceleration automation trigger available to anyone using any type of AAPS algo.

Kenoubi, 2023-08-06:
You can do this the super ugly way: short avg delta > 3 && long avg delta >5 && short avg delta < 5 && long avg delta < 7, then make a half dozen rules like that...
(If you put your rules in the right order you can probably leave off the upper bounds.  But yes, this is pretty much as ugly as it sounds, and being able to do simple arithmetic in automation somehow would make it *way* more powerful.  Might also tempt people to do things that are hard to do reliably, but IMO such people (e.g. me) are going to try to do them anyway...)