nightscout/AndroidAPS #2513: Commit for "Close gatt when BLE is disconnected" deteriorates connection reliability

vanelsberg, 2023-04-21:
**AndroidAPS 3.2.0-dev-i**
**Build**: 5e2614f9f6
**Flavor**: fullrelease / Engineering mode
Pump: DASH
Phone: Samsung S22 Ultra

**Problem description:**

As a result of this change APPS connects faster to the Pod but connection reliability has become a problem.
Getting timeouts on connecting a few time a day.

Connection connection time has no relevance but connection reliability is first priority. Also on most supported phones there is no problem with connecting to the pod. Current BLE implementation was field-tested for a long time on many phones and with many DASH versions.

I do not see any reason to change, please do not fix what is working for most phones!?
So suggest undoing this change and reconsider necessity.

[image]
[image]

[AndroidAPS_LOG_1682057146032.log.zip](https://github.com/nightscout/AndroidAPS/files/11292807/AndroidAPS_LOG_1682057146032.log.zip)

jbr7rr, 2023-04-21:
This fixes the issue mentioned:
https://github.com/nightscout/AndroidAPS/pull/2514

And I would argue that the reliability was not always there, on most pixel phones and some samsungs, one had to enable the bluetooth watchdog, and always had a sub optimal success rate on connection attempt success (90-98%)

vanelsberg, 2023-04-21:
@jbr7rr  Thanks. Quite busy atm, but will test this a.s.a.p.
Pixel has been problematic from the start and was concluded it was "problematic" when it comes to "DASH compatibility".

Let see if you can improve DASH's BLE implementation and discuss how to test your final code on as many hardware hardware as possible. I'll DM you!?

vanelsberg, 2023-04-27:
Problem solved with https://github.com/nightscout/AndroidAPS/commit/9c902fb141b8c784e1ac245fc35d3577113827d7

closed 2023-04-27 by vanelsberg