nightscout/AndroidAPS #2517: How does Total Delivered calculated for Omnipod DASH

birdfly, 2023-04-22:
Reporting bugs

Backgroud: 

----Hello, my AndroidAPS cannot automatically run the closed-loop function. The blood glucose data is constantly updated and synchronized, and the communication with the insulin pump is normal. However, I can see a progress bar scrolling at the bottom of the home page, as if it is loading something.

In addition, there seems to be a problem with my Nightscout upload. The upload queue often remains stagnant, and the logs show that data from some dates are being uploaded repeatedly. Moreover, when I enable Nightscout upload, the usage time of my insulin pump and CGM becomes incorrect, including carbohydrates. It seems that AndroidAPS does not prioritize the latest calculation tasks and local device data, and is stuck in a loop with some historical data.

I had sent logs via email
------
Yesterday I just reset my database before I activate a new DASH Pod, then changed a new Pod and took 2 boluses(0.5u and 1.0 U),but I saw that the Omnipod Page displayed 【Total Delivered ：4.65 U】, so I went to the Pod history page to check ,there are only two boluses events :0.5u and 1u ,so I am curious about how the Total Delivered calculated for Omnipod DASH .

here I attached two screenshots ：

[image]
 
[image]

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).

----Build: 3.1.0.3  Base Code 
----Branch:  https://github.com/tim2000s/Boost-master-v3/tree/Boost_Test_Platform

- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

mtanjagithub, 2023-04-22:
It is normal for the Dash to fill the cannula by starting up. The Dash needs +/- 3 U for this

birdfly, 2023-04-22:
> It is normal for the Dash to fill the cannula by starting up. The Dash needs +/- 3 U for this

Understood, but as I know the Boost algorithm uses TDD to calculate ISF

vanelsberg, 2023-04-24:
https://github.com/nightscout/AndroidAPS/issues/805

vanelsberg, 2023-04-24:
> but as I know the Boost algorithm uses TDD to calculate ISF
TTD is calculated from data in the database,

If Boost would use the "Insuline Delivered" as reported by DASH to calculate it would fail:
- DASH can not report insuline reservoir values above 50U (it will show report 50U until below)
- The reservoir value does not take into account insulin registered as "record only" (like when using a PEN)

closed 2023-08-21 by MilosKozak