nightscout/AndroidAPS #2523: Glucomen daypump patch activation problem

Marcovdh93, 2023-04-28:
I’m new here but i think it’s good to share with u:

for 2 months I’m using aaps with the Glucogen daypump.
When I’m connecting a new patch expediently my screen switched from vertical to horizontal. What happened was that de app started the proces of connecting again. 
I wasn’t able to reconnect with the patch so I needed to pick up a new one. 

So in short terms. The proces of connecting starts over when the screen rotates. This could be nice to be fixen.

Is this something that can be fixed? Or is it only with my type of phone?

I’m using aap 3.2.0.
Having a Motorola e22

mtanjagithub, 2023-04-28:
I have the same problem. I have to wrap my old pod and paired watch in aluminum foil to prevent them from interfering with the BT connection. Then you'll be fine without losing a pod. 

I am using Galaxy Watch 5 pro 
Glucomen day pump with AAPS 3.2.0-DEV-i 39dd25af07 on Cubot Kingkong mini 2 pro

I have not had this problem while using the Omnipod Dash with the same setup. I've had the Dash for 3 months

fjpezuela, 2023-05-01:
Good morning. The same thing happened to me. I updated the AAPS version to the latest dev version available and started the patch change. I was able to correctly discard the previously active patch, but when I tried to activate the new one, it entered a loop and couldn't get past step 1. I switched back to the previous version of AAPS, but it was still stuck in the loop and I eventually lost the patch. I activated the next patch using the previous version I had, and everything worked correctly. I haven't dared to update again since I'm not sure if the problem is resolved, and I have few PODs.

eduardomarron, 2023-05-10:
Same problem here, 3 pods worked like a charm but since yesterday it is not possible to pair the pods again. it gets stuck trying to connect and nothing happens.

MilosKozak, 2023-05-14:
resolved by https://github.com/nightscout/AndroidAPS/commit/fba804b2cb29e7ca25bad43a696bd77e1bf54fdf

closed 2023-05-14 by MilosKozak

eduardomarron, 2023-05-17:
Hello there,
Still the same problem, this is the order I followed:

1.- Update AAPS to 320 DEV-J build: c86edb94fd-2023.05.14
2.- Deactivate the previous pod in AAPS
3.- Link the new one. Pair identifier was shown twice as expected. Clicked OK.
4.- AAPS gets stuck in "Connecting" and nothing happens.

Then I tried

Remove AAPS cache, didn't work.
Removed all data and restarted aaps configuration -> didn't work

Using Samsung A52s. Android 13.

MilosKozak, 2023-05-17:
without logs it's impossible to do anything

eduardomarron, 2023-05-17:
attached log. Change was done yesterday 2023.05.16 around 21.30 aprox. Hope it helps. Thanks a lot.
[AndroidAPS._2023-05-16_21-46-11_.3.zip](https://github.com/nightscout/AndroidAPS/files/11498863/AndroidAPS._2023-05-16_21-46-11_.3.zip)

MilosKozak, 2023-05-18:
it seems aaps crashed but nothing is logged. sorry

MilosKozak, 2023-05-18:
I found this on another place:
```
21:44:12.098 [RxComputationThreadPool-6] I/PUMP: [EopatchViewModel.updateSetupStep():597]: curSetupStep: SCAN_STARTED, newSetupStep: BONDING_STARTED
21:44:12.100 [RxComputationThreadPool-6] D/PUMPCOMM: [TaskQueue.publishTicket():53]: publishTicket() Task #:17 is assigned func:START_BOND
21:44:12.102 [RxComputationThreadPool-6] D/PUMPCOMM: [TaskQueue.lambda$isReady$4$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-TaskQueue():40]: Task #:17 started     func:START_BOND
21:44:13.536 [RxCachedThreadScheduler-44] D/AUTOMATION: [AutomationPlugin$onStart$11.accept():176]: Grabbed new BT event: EventBTChange[deviceAddress=F3:1B:D5:9B:C4:FA,deviceName=EO30-C4FA#0.2.41,state=CONNECT]
21:44:13.548 [RxCachedThreadScheduler-44] D/AUTOMATION: [AutomationPlugin.processActions$automation_fullRelease():250]: processActions
21:44:13.550 [RxCachedThreadScheduler-44] D/AUTOMATION: [TriggerBg.shouldRun():63]: NOT ready for execution: Glucose is greater than 160 MGDL
21:44:13.552 [RxCachedThreadScheduler-44] D/AUTOMATION: [TriggerBg.shouldRun():63]: NOT ready for execution: Glucose is lesser than 85 MGDL
21:44:14.044 [RxBleThread-1] D/PUMP: [PatchManagerImpl.lambda$onInit$4$info-nightscout-androidaps-plugins-pump-eopatch-ble-PatchManagerImpl():169]: Has the date or time changed? false
21:44:14.071 [main] D/CORE: [DummyService.onStartCommand():68]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0x0 flags=0xa color=0x00000000 category=status vis=PRIVATE semFlags=0x0 semPriority=0 semMissedCount=0)
21:44:14.366 [RxSingleScheduler-1] D/PUMPCOMM: [TaskQueue.done():73]: done() Task #:17 completed   func:START_BOND  task remaining:0
21:44:14.367 [RxSingleScheduler-1] E/PUMPCOMM: [StartBondTask.lambda$start$6$info-nightscout-androidaps-plugins-pump-eopatch-ble-task-StartBondTask():43]: StartBondTask error
21:44:14.367 [RxSingleScheduler-1] I/PUMP: [EopatchViewModel.updateSetupStep():597]: curSetupStep: BONDING_STARTED, newSetupStep: BONDING_FAILED
21:44:14.369 [RxSingleScheduler-1] E/PUMP: [SingleExtensionKt$subscribeDefault$1.accept():12]: onError
java.lang.Exception: null
	at info.nightscout.androidaps.plugins.pump.eopatch.ble.task.StartBondTask.lambda$start$2(StartBondTask.java:35)
	at info.nightscout.androidaps.plugins.pump.eopatch.ble.task.StartBondTask$$ExternalSyntheticLambda3.accept(Unknown Source:2)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext(ObservableDoOnEach.java:93)
	at io.reactivex.rxjava3.observers.SerializedObserver.onNext(SerializedObserver.java:114)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$SourceObserver$InnerObserver.onNext(ObservableConcatMap.java:252)
	at io.reactivex.rxjava3.internal.util.NotificationLite.accept(NotificationLite.java:247)
	at io.reactivex.rxjava3.subjects.BehaviorSubject$BehaviorDisposable.test(BehaviorSubject.java:507)
	at io.reactivex.rxjava3.subjects.BehaviorSubject$BehaviorDisposable.emitFirst(BehaviorSubject.java:468)
	at io.reactivex.rxjava3.subjects.BehaviorSubject.subscribeActual(BehaviorSubject.java:224)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$SourceObserver.drain(ObservableConcatMap.java:223)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$SourceObserver.onNext(ObservableConcatMap.java:132)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext(ObservableDoOnEach.java:101)
	at io.reactivex.rxjava3.internal.operators.mixed.ObservableConcatMapSingle$ConcatMapSingleMainObserver.drain(ObservableConcatMapSingle.java:201)
	at io.reactivex.rxjava3.internal.operators.mixed.ObservableConcatMapSingle$ConcatMapSingleMainObserver.innerSuccess(ObservableConcatMapSingle.java:96)
	at io.reactivex.rxjava3.internal.operators.mixed.ObservableConcatMapSingle$ConcatMapSingleMainObserver$ConcatMapSingleObserver.onSuccess(ObservableConcatMapSingle.java:235)
	at io.reactivex.rxjava3.internal.operators.single.SingleDoOnSuccess$DoOnSuccess.onSuccess(SingleDoOnSuccess.java:60)
	at io.reactivex.rxjava3.internal.operators.single.SingleObserveOn$ObserveOnSingleObserver.run(SingleObserveOn.java:81)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:25)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)
21:44:15.309 [main] D/PUMP: [PatchStateManager.lambda$updatePatchState$2$info-nightscout-androidaps-plugins-pump-eopatch-ble-PatchStateManager():46]: PatchState GlobalTime:1970/01/01 01:02:16 -->  87  00  02  00  A9  01  00  00  00  04  B0  3C  NeedPriming:On NeedNeedleSensing:On [NormalBasal] Reg:    Act:    Paused:   remainedInsulin:60  remainedPumpCycle:1200(60.0)  battery:3,14V
```
this is related to android pairing problem

eduardomarron, 2023-05-18:
> this is related to android pairing problem. Did you respond to pairing dialog properly and asap?

Yes, I clicked OK twice when AAPS prompted me to pair.

mtanjagithub, 2023-05-20:
After updating to 3.2J, the problem has not been solved for me either.

AAPS 3.2.0-dev-j
Build: 0a63a59bb6-2023.05.17
Flavor: fullrelease
Engineering modus is geactiveerd
[AndroidAPS._2023-05-20_02-00-04_.zip](https://github.com/nightscout/AndroidAPS/files/11521817/AndroidAPS._2023-05-20_02-00-04_.zip)

time 08:42:26.046

mtanjagithub, 2023-05-23:
I have an update. Last weeks I was back on the Omnipod Dash because the Glucomen day wouldn't pair (these were the same problems as described above). Downloaded the original Narsha app for Glucomen yesterday. During the installation, you were asked for various permissions such as: Location and battery. But I also received the message that to set the date and time 'use times offered by network' had to be set. I have adjusted this setting and (after restarting the phone just to be sure) I was able to successfully start a new Glucomen Day pump.

I have no idea if this setting caused all the problems or if the problem is definitely over. But for the next 3.5 days I am happy with the working pump via AAPS

parapenT1sta, 2023-05-23:
> 

Librelink app for Abbott Libre also ask for that setting. Was a great Idea for AAPS check for that setting at every start to avoid similar errors. 

@MilosKozak 

[image]

eduardomarron, 2023-05-24:
> I have an update. Last weeks I was back on the Omnipod Dash because the Glucomen day wouldn't pair (these were the same problems as described above). Downloaded the original Narsha app for Glucomen [...]

Hello,

I'm afraid it worked becuase you changed from Omnipod Dash to Glucomen. it works for me as well when:
1.- After pairing problems I move back to Roche Insight
2.- Use the insight for 3 days.
3.- Activate again the glucomen.

The real test is when you deactivate the old pod and activate the new one. (I have the hour/time set to use the network). Let us know if it works for you.

Marcovdh93, 2023-05-24:
I started this topic because of the rotating screen. At this moment i solved it to put my phone on "not auto-rotating" And secure my phone by not picking it up during the connecting proces. 

So today there came up an other problem. 
After the proces of connecting a new pod the screen stayed on "Connecting" (dutch "verbinden"). After closing and opening the app the pod did work but:
- Time is still counting: 3,5 + days.
- It gives insuline but doesn't registrate it in the app date (you find 0,32e in picture, that number should be higher).
- the error is dutch and still shows dutch when i switch language but means: "Data comes from other pump. Change pump driver to reset the status of the pump".

I did download logs en added them to this post.

Does anyone know about this issui and any recommondations on avoiding this problem?  

Im still on:
I’m using aap 3.2.0.
Having a Motorola e22

[AndroidAPS_LOG_1684916973865.log.zip](https://github.com/nightscout/AndroidAPS/files/11552193/AndroidAPS_LOG_1684916973865.log.zip)

[image]

mtanjagithub, 2023-05-24:
> Hello,
>
> I'm afraid it worked becuase you changed from Omnipod Dash to Glucomen. it works for me as well when: 1.- After pairing problems I move back to Roche Insight 2.- Use the insight for 3 days. [...]

I've noticed the same thing with previous problems. That after using the Dash that the pairing works again with the Glucomen.

@MilosKozak  could there be an error in disconnecting the old Glucomen patch? Could it be that AAPS is going through the processes too quickly? or being shut down too soon so that the BT connection with the old patch is not completely stopped? could a solution be to add a pause in the discnetting to give the system more time to complete processes and boot up? When creating VBA, adding pauses between processes can prevent/fix problems.  Please reopen the issue

mtanjagithub, 2023-06-14:
> Hello,
>
> I'm afraid it worked becuase you changed from Omnipod Dash to Glucomen. it works for me as well when: 1.- After pairing problems I move back to Roche Insight 2.- Use the insight for 3 days. [...]

After updating I still experience the same thing
What I notice is that after updating or reinstalling I am able to pair new EOpatch2 twice. If I don't update or reinstall AAPS after that, I'll have to switch from the EOpatch2 to the Dash after 3.5 days. After the 3 days I can reconnect the EOpatch again. If I don't do this, the attachment of the EOpatch2 hangs at step 2, even after confirming the two requests. 

@MilosKozak , @fjpezuela , Could this be related to #2573?  because it seems that after reinstalling or updating AAPS everything seems to work fine for a few days and then the errors starts 

AAPS 3.2.0-dev-j
Build: cd9f5e4723-2023.06.01
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
phone Cubolt Kingkong mini pro 2

Disconnect Dash and tried to connect new EOPatch2: failed to
21:06:44.800 [main] D/UI: [ActivityMonitor.onActivityResumed():50]: onActivityResumed: EopatchActivity

After I updated AAPS. Just restarted the existing APK file I already had and updated AAPS with the same version (not deleted first). Simply put the same version on top of each other. After that I was able to activate a new EOPatch2 pump. at 21:24:41.898 [RxSingleScheduler-1] D/PUMP: [PumpSyncImplementation.confirmActivePump():105]: Registering new pump Eoflow Eopatch2 J5G5A022498 (new pump attached)

[AndroidAPS._2023-06-12_21-19-38_ (2 ).txt](https://github.com/nightscout/AndroidAPS/files/11747978/AndroidAPS._2023-06-12_21-19-38_.2.txt)
[AndroidAPS._2023-06-12_21-19-38_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747980/AndroidAPS._2023-06-12_21-19-38_.2.txt)
[AndroidAPS._2023-06-12_21-14-03_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747981/AndroidAPS._2023-06-12_21-14-03_.2.txt)
[AndroidAPS._2023-06-12_21-15-03_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747982/AndroidAPS._2023-06-12_21-15-03_.2.txt)

fjpezuela, 2023-06-17:
> > > I have an update. Last weeks I was back on the Omnipod Dash because the Glucomen day wouldn't pair (these were the same problems as described above). Downloaded the original Narsha app for Glucomen yesterday. During the installation, you were asked for various permissions such as: Location and battery. But I also received the message that to set the date and time 'use times offered by network' had to be set. I have adjusted this setting and (after restarting the phone just to be sure) I was able to successfully start a new Glucomen Day pump.
> > > I have no idea if this setting caused all the problems or if the problem is definitely over. But for the next 3.5 days I am happy with the working pump via AAPS
> > 
> > 
> > Hello,
> > I'm afraid it worked becuase you changed from Omnipod Dash to Glucomen. it works for me as well when: 1.- After pairing problems I move back to Roche Insight 2.- Use the insight for 3 days. 3.- Activate again the glucomen.
> > The real test is when you deactivate the old pod and activate the new one. (I have the hour/time set to use the network). Let us know if it works for you.
> 
> After updating I still experience the same thing What I notice is that after updating or reinstalling I am able to pair new EOpatch2 twice. If I don't update or reinstall AAPS after that, I'll have to switch from the EOpatch2 to the Dash after 3.5 days. After the 3 days I can reconnect the EOpatch again. If I don't do this, the attachment of the EOpatch2 hangs at step 2, even after confirming the two requests.
> 
> @MilosKozak , @fjpezuela , Could this be related to #2573? because it seems that after reinstalling or updating AAPS everything seems to work fine for a few days and then the errors starts
> 
> AAPS 3.2.0-dev-j Build: [cd9f5e4](https://github.com/nightscout/AndroidAPS/commit/cd9f5e472396786427a1538115b661ced167bd92)-2023.06.01 Remote: https://github.com/nightscout/AndroidAPS Flavor: fullrelease phone Cubolt Kingkong mini pro 2
> 
> Disconnect Dash and tried to connect new EOPatch2: failed to 21:06:44.800 [main] D/UI: [ActivityMonitor.onActivityResumed():50]: onActivityResumed: EopatchActivity
> 
> After I updated AAPS. Just restarted the existing APK file I already had and updated AAPS with the same version (not deleted first). Simply put the same version on top of each other. After that I was able to activate a new EOPatch2 pump. at 21:24:41.898 [RxSingleScheduler-1] D/PUMP: [PumpSyncImplementation.confirmActivePump():105]: Registering new pump Eoflow Eopatch2 J5G5A022498 (new pump attached)
> 
> [AndroidAPS._2023-06-12_21-19-38_ (2 ).txt](https://github.com/nightscout/AndroidAPS/files/11747978/AndroidAPS._2023-06-12_21-19-38_.2.txt) [AndroidAPS._2023-06-12_21-19-38_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747980/AndroidAPS._2023-06-12_21-19-38_.2.txt) [AndroidAPS._2023-06-12_21-14-03_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747981/AndroidAPS._2023-06-12_21-14-03_.2.txt) [AndroidAPS._2023-06-12_21-15-03_ (2).txt](https://github.com/nightscout/AndroidAPS/files/11747982/AndroidAPS._2023-06-12_21-15-03_.2.txt)

Hello @mtanjagithub,

I tried installing the fix a5774b6e2eb398311c80e9e00c01bede82978633, where the problem is fixed when opening preferences after patch activation (see ticket: https://github.com/nightscout/AndroidAPS/issues/2573), but I have gone back, as I noticed the following problems:

- Notifications do not appear (Messages showing that the patch is going to expire in x hours, that the patch has expired, etc, now do not appear).

- The continuous patch change warning beeps have reappeared (before the fix it sounded at 4 times: at the reminder before 3.5 days, at 3.5 days, at the reminder before 4 days and at 4 days).

These two points are important to me, so I prefer the version with the problems when opening the preferences, because by switching to "virtual pump" I can open them, make the changes I need and then re-select EOPatch2.

Is this fix temporary to fix the preferences problem, can you re-display the notifications and remove the continuous beeps for patch expiration?

Thank you very much in advance.

mtanjagithub, 2023-06-17:
> I tried installing the fix [a5774b6](https://github.com/nightscout/AndroidAPS/commit/a5774b6e2eb398311c80e9e00c01bede82978633), where the problem is fixed when opening preferences after patch activation (see ticket: #2573), but I have gone back, as I noticed the following problems:
> 
> * Notifications do not appear (Messages showing that the patch is going to expire in x hours, that the patch has expired, etc, now do not appear).
> * The continuous patch change warning beeps have reappeared (before the fix it sounded at 4 times: at the reminder before 3.5 days, at 3.5 days, at the reminder before 4 days and at 4 days).
> 
> These two points are important to me, so I prefer the version with the problems when opening the preferences, because by switching to "virtual pump" I can open them, make the changes I need and then re-select EOPatch2.
> 
> Is this fix temporary to fix the preferences problem, can you re-display the notifications and remove the continuous beeps for patch expiration?
> 
> Thank you very much in advance.

Hey,

Last night I had changed a EOPatch2 pod again. Unfortunately, reconnecting a new EOPatch2 failed. I have taken the steps as described above, but without succes. Now back on the Dash again.

I do have access to the preferences. I don't have any Notifications and I can't turn off the warning beeps either. This was with an APK file built on 13-04-2023. This morning I built a new APK.

fjpezuela, 2023-06-17:
> Hey,
>
> Last night I had changed a EOPatch2 pod again. Unfortunately, reconnecting a new EOPatch2 failed. I have taken the steps as described above, but without succes. Now back on the Dash again.
>
> I [...]

Hello,

Thank you for your reply. Just in case it helps. 

I could not see the notification message informing of the patch expiration and that you need to confirm, so I was unable to do so. 

is it possible that not being able to confirm that notification caused the beeps? 

The beeps seemed to be different than the usual expiration beeps, although I was in a place with quite a bit of background noise and can't say for sure.

fjpezuela, 2023-06-20:
> Hey,
>
> Last night I had changed a EOPatch2 pod again. Unfortunately, reconnecting a new EOPatch2 failed. I have taken the steps as described above, but without succes. Now back on the Dash again.
>
> I [...]

Hello again,

I updated to the latest version (which includes the EOPATCH2 fix) and this time I did get the patch expiration notification and I was able to confirm it. I think the problem with the previous patch was that I didn't get to see the notification or somehow didn't confirm it (I was in a noisy place and didn't notice).

Do you know how much time there is to confirm the patch expiration notification? Could it be self-confirmed after x time?

Thank you so much.

mtanjagithub, 2023-06-20:
> Hello again,
>
> I updated to the latest version (which includes the EOPATCH2 fix) and this time I did get the patch expiration notification and I was able to confirm it. I think the problem with the [...]

Hello again,
3,5 dag + 12 hours i think